dji-sdk/Onboard-SDK-ROS #210: Regarding /dji_sdk/rtk_info_position topic

qboticslabs, 2018-06-07:
Hello

We have some confusion regarding the /dji_sdk/rtk_info_position (std_msgs/UInt8). 
From the http://wiki.ros.org/dji_sdk, it is given that

"The type of RTK positioning, indicating different solutions for calculating the position published at 10hz. Only available when RTK present". 

Can you mention different kinds of RTK positioning, because we can't decode it from the integer value?

 Also please let us know if there is a topic to detect the RTK is in sync with the Ground station or not. I think it will be great if you can add that topic to the ROS interface.

Regards
Lentin Joseph

rohitsantdji, 2018-06-12:
Please see the documentation in the [API reference](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Telemetry_1_1RTK.html)  - look for the posHealthFlag documentation, which is the non-ROS equivalent for rtk_info_position.

qboticslabs, 2018-06-12:
Thanks for that

closed 2018-06-12 by qboticslabs