dji-sdk/Onboard-SDK-ROS #212: Drone task control problem

venenokk, 2018-06-11:
Software: `Ubuntu 14.04`, `ROS(Indigo)`, `DJI_ONBOARD_SDK_ROS(V3.6)`
Hardware: Only `A3`(That means I didn't assemble a drone)
Problem: The response result of service "drone_task_control' is always false. 
Specifically, the output is
<pre>
SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: removed
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: removed
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [17513]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 177f9d40-6d70-11e8-bf62-0cc47acc49e8
process[rosout-1]: started with pid [17526]
started core service [/rosout]
process[dji_sdk-2]: started with pid [17529]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.
[ INFO] [1528718883.731376545]: Succeeded to read from serial device

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L687: Device Serial No. = 0670140957

STATUS/1 @ parseDroneVersionInfo, L689: Hardware = A3

STATUS/1 @ parseDroneVersionInfo, L690: Firmware = 3.2.36.8

ERRORLOG/1 @ initVirtualRC, L1029: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1239: version 0x3022408

STATUS/1 @ activate, L1252: Activation successful

STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ initGimbal, L850: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ initGimbal, L894: Gimbal not mounted!
[ INFO] [1528718890.028945513]: drone activated
[ INFO] [1528718890.100215204]: Use data subscription to get telemetry data!
[ INFO] [1528718890.100261039]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L237: Verify subscription successful.
STATUS/1 @ startPackage, L343: Start package 2 result: 0.
STATUS/1 @ startPackage, L345: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L343: Start package 1 result: 0.
STATUS/1 @ startPackage, L345: Package 1 info: freq=50, nTopics=10.

STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=10, nTopics=1.

STATUS/1 @ removePackage, L461: Remove package 0 successful.
STATUS/1 @ startPackage, L343: Start package 0 result: 0.
STATUS/1 @ startPackage, L345: Package 0 info: freq=10, nTopics=11.

STATUS/1 @ startPackage, L343: Start package 3 result: 0.
STATUS/1 @ startPackage, L345: Package 3 info: freq=400, nTopics=1.

STATUS/1 @ getErrorCodeMessage, L644: droneTaskCallback
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_MOTOR_ON
</pre>
So as you see, the last two rows of the output are
<pre>
STATUS/1 @ getErrorCodeMessage, L644: droneTaskCallback
STATUS/1 @ getCMDIDTaskMSG, L824: CONTROLLER_MOTOR_ON
</pre>
Also, the raw output of `rosservice call /dji_sdk/drone_task_control 4` is
<pre>
rosservice call /dji_sdk/drone_task_control 4
result: False
cmd_set: 1
cmd_id: 1
ack_data: 1
</pre>
There're similar problems posted by other users. But maybe that's not my case?
Can anyone help? 
Thanks.

venenokk, 2018-06-12:
Refer to this question on stackoverflow [https://stackoverflow.com/questions/48494194/drone-wont-take-of-nor-spin-its-motor-with-onboard-sdk-sample-code](url) It's said as a calibration problem of compass.
So to verify the solution, I installed the compass and calibrated. After that, yeah it succeeded to implement the service.
However, a new problem that "Takeoff failed. Motors are not spinning" raised. But now I consider that only using A3 isn't a good way. Perhaps one needs assemble a drone and finish all calibration and at least, can fly the drone with remote controller after that try ROS control.

closed 2018-06-12 by venenokk

AbhishekGS, 2019-08-12:
Hi All, @venenokk ,

I am facing a similar problem as you have described above. My details are as below:
Software: Ubuntu 16.04.5 (LTS), ROS(Kinetic), DJI_ONBOARD_SDK_ROS(V3.8)
Hardware: Only N3(That means I didn't assemble a drone, and not connected Compass as well)
                 Futaba T6K Transmitter and R3006SB SBUS receiver
Problem: The response result of service "drone_task_control' is always false.
Specifically, the output is

On Terminal 1:

```
asheshad@ubuntu:~/dji_catkin_ws/src/Onboard-SDK-ROS/dji_sdk$ roslaunch dji_sdk sdk.launch
... logging to /home/asheshad/.ros/log/f8b721d0-bcf1-11e9-a254-000c292328c9/roslaunch-ubuntu-37369.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38529/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1072802
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: 491b3222869b48821...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [37379]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8b721d0-bcf1-11e9-a254-000c292328c9
process[rosout-1]: started with pid [37392]
started core service [/rosout]
process[dji_sdk-2]: started with pid [37403]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0DBDF5E0020175

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x303082F

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
[ INFO] [1565608313.263188915]: drone activated
[ INFO] [1565608313.381370117]: Use data subscription to get telemetry data!
[ INFO] [1565608313.381464771]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 2 result: 0.
STATUS/1 @ startPackage, L352: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L350: Start package 1 result: 0.
STATUS/1 @ startPackage, L352: Package 1 info: freq=50, nTopics=14.

STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=5, nTopics=6.

STATUS/1 @ startPackage, L350: Start package 3 result: 0.
STATUS/1 @ startPackage, L352: Package 3 info: freq=400, nTopics=1.

STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

```

The last 2 rows of the output are:
```
STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

```

```
asheshad@ubuntu:~/dji_catkin_ws$ rosservice call /dji_sdk/drone_task_control "task: 4"
result: False
cmd_set: 1
cmd_id: 1
ack_data: 113
```

Could you please let me know what worked for you?

Thanks,
Abhishek