dji-sdk/Onboard-SDK-ROS #215: Onboard SDK stop sending data after approximately 30 seconds

Gerjen100, 2018-06-15:
Hello,

Today i was experimenting with the onboard SDK on a raspberry pi 3 and i noticed that the sdk stopped sending any data to the rosmaster after about 30 seconds. The master itself did not die as other packages kept sending their data. There was no notice coming from the SDK that something was wrong. The setup i am using is a DJI M100 with a FTDI323 USB-TTL converter. I am using this same setup in combination with a intel nuc instead of a raspberry pi 3, there i do not encounter any issues.

I have done some CPU/memory monitoring and while running the onboard sdk the cpu and memory are not even at half their stress capacity. 

Thanks in advance, Gerjen.

L4ncelot, 2019-02-19:
I'm experiencing the same problem on raspberry pi 3 as you did. Have you ever managed to solve the issue?

Gerjen100, 2019-02-19:
Hi,

Yes i have, a re-install of ubuntu seemed to resolve the issue. I have seen something similar on another drone i was using, there we resolved the issue by using a different FTDI chip.

L4ncelot, 2019-02-19:
Different FTDI chip didn't help, I'll try new os tomorrow. This issue is really strange. But thanks anyway.

closed 2019-02-27 by Gerjen100

L4ncelot, 2019-02-27:
So it was in the end bad FTDI chip indeed. We used some FTDIs from china that can't just handle those high baudrates like 921600. After changing to FTDI chip from america everything works just fine. We also checked the autopilot on our oscilloscope and it works just fine.

lanyusea, 2019-02-28:
good catch