dji-sdk/Onboard-SDK-ROS #216: Setup Camera Stream not available as a rosservice

mikeklinker129, 2018-06-15:
With version 3.6, when you do a 
```
catkin_make
source devel/setup.bash
rosservice list
```
The service to set up a camera stream is not available. 

Current output:
```
:~/DJI/catkin_ws$ rosservice list
/dji_sdk/activation
/dji_sdk/camera_action
/dji_sdk/drone_arm_control
/dji_sdk/drone_task_control
/dji_sdk/get_loggers
/dji_sdk/mfio_config
/dji_sdk/mfio_set_value
/dji_sdk/mission_hotpoint_action
/dji_sdk/mission_hotpoint_getInfo
/dji_sdk/mission_hotpoint_resetYaw
/dji_sdk/mission_hotpoint_updateRadius
/dji_sdk/mission_hotpoint_updateYawRate
/dji_sdk/mission_hotpoint_upload
/dji_sdk/mission_status
/dji_sdk/mission_waypoint_action
/dji_sdk/mission_waypoint_getInfo
/dji_sdk/mission_waypoint_getSpeed
/dji_sdk/mission_waypoint_setSpeed
/dji_sdk/mission_waypoint_upload
/dji_sdk/query_drone_version
/dji_sdk/sdk_control_authority
/dji_sdk/send_data_to_mobile
/dji_sdk/set_hardsyc
/dji_sdk/set_local_pos_ref
/dji_sdk/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level

```

rcabg, 2018-06-16:
Hey @mikeklinker129 ,

did you compile DJI Onboard SDK with the flag ADVANCED_SENSING set to ON? [Link](https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#linux-oes)

Also, can you describe your hardware setup please?

mikeklinker129, 2018-06-18:
Using an M210 connected via FTDI cable to my laptop. 
Advanced sensing breaks my build... see #337

closed 2021-07-07 by dji-kyle