dji-sdk/Onboard-SDK-ROS #218: Switching Flight Control Modeswhile flying

reygago, 2018-06-21:
Hi,
I'm currently developing an indoor solution for Matrice 100 using Guidance - so, I'm sending commands with the flag  HORIZONTAL_VELOCITY. This works fine, while I'm receiving velocity input from guidace. If light changes or guidace stops detecting features (velocity = 0.0) the, as expected, the flight command stops working and the drone starts drifitng. If guidance restores, then the drone resume his task.
To avoid this problem, I'm trying to change the flag to HORIZONTAL_ANGLE and send commands as roll and pitch for a while, until guidance is restored.
I'm having two problems here: 
1) I have to estimate velocity with other sensors and design an angle based control - this takes time and if someone can help i is appreciated
2) when I return the flag to HORIZONTAL_VELOCITY, the drone gets crazy and does erratic movement for about 2-3 seconds before stabilizing again. Is there a way to make a smooth change? Is this a bug?

Thanks in advance,
Reynaldo

reygago, 2018-07-11:
Some additional info for this topic:
If I switch from HORIZONTAL_ANGLE to HORIZONTAL_VELOCITY while the drone is holding still (command roll and pitch = 0.0), there is no problem while switching. The erratic movement happens only when the thedrone is moving.

dangviethieu-hntv, 2018-08-14:
@reygago Oh I also have problem like your. But I get velocity input from guidance **incorrect** (velocity#0.0 when drone stops). Can you help me share a little experience about setting guidance. Please !!! Thanks you very much.

reygago, 2018-08-14:
Ok... About the Guidance system:
The velocity that the drone uses to control (receive commands with horizontal_velocity flag) is the one in dji_sdk/velocity topic. This velocity is a result of sensor fusion, probably it uses guidance velocity, IMU and GPS info, plus some "blackbox" info about the drone flight model - in my case, I have no GPS signal, so the sensor fusion is only based on Guidance and IMU. After guidance topics equals zero, it takes 2 seconds to dji_sdk/velocity goes to zero.So, the drone stops receiving commands  when this topic (dji_sdk/velocity) equals zero, not the /guidance/velocity topic (this is based on many tests that we made) -  Usuallly, only x and y vector are equal zero, z usually keeps estimating.

Then, when guidance system restores (/guidance/velocity !=0), it takes approximately 2 seconds to dji_sdk/velocity restores - in my case, few seconds without velocity makes the drone crash. When dji_sdk/velocity is diferent than zero again, it is possible to resume sending velocity commands.

My problem is that, when guidance fails, my drone is in autonomous mode and need to keep going foward. So, when I detect a failure, I switch to angle control (and it took me a while to build this controller) and keep on it until the drone stops. Then, I switch back to velocity control and resume the task. I have to wait until it stops (command roll and pitch = 0) to switch flags, or else it would start making an erratic movement

dangviethieu-hntv, 2018-08-15:
@reygago I am most grateful...You've made my day !!!
I really thought simple about this problem. I only get 1 parameter is velocity of guidance and it alway unstable, I will research more ...
Thanks you very much !!! I hope you can solving yourself problem.

dangviethieu-hntv, 2018-08-23:
@reygago why we don't use velocity parameter get from "**dji_sdk/velocity**" topic, it is more exactly.

reygago, 2018-08-31:
I don't remember where I found this reference, but I transformed all velocities on body orientation. So, using the quaternion read on dji_sdk/imu, transforming the dji_sdk/velocity like this
        velocity_m100 = tf2::quatRotate(tf2::Quaternion(_orientation.x,_orientation.y, _orientation.z, _orientation.w).inverse(),tf2::Vector3(msg->vector.x,msg->vector.y, msg->vector.z));

and the transformation of guidance velocity like the following
velocity_guidance = tf2::quatRotate(tf2::Quaternion(_orientation.x,_orientation.y, _orientation.z, _orientation.w).inverse(),tf2::Vector3(msg->vector.y,msg->vector.x, -msg->vector.z));

This is what I'm using on my code - if mym memory is right, drone velocity reference is NED and guidance is ENU.

closed 2021-07-07 by dji-kyle