dji-sdk/Onboard-SDK-ROS #220: Can I use Onboard-SDK-Ros for DJI Spark to get the video feed to my Linux PC?

mohansriramakrishna, 2018-07-12:
I just want to know is there any way to get the live feed from DJI spark to my PC either by using OpenCv or is there any other way.

rohitsantdji, 2018-07-18:
Hi @mohansriramakrishna , Onboard SDK does not support Spark. However you can get the live feed on Android/iOS using Mobile SDK and use OpenCV on those platforms.

closed 2018-07-18 by rohitsantdji

Leush19, 2018-07-18:
Hey! I'm really new into this but I'm looking for the DJI Spark for development purposes and research. 
I wonder if threre's any compatibility with ROS using the Spark. I've done some research and I know that there´s onboard support with ROS but not sure about the mobile SDK.
Also I'm considering the bebop 2 as I know that it does support compatibility with ROS and OpenCV
though I'm really into the spark and I wish it would have same compatibility.
Sorry about my English, not native speaker.
Thank's !!