dji-sdk/Onboard-SDK-ROS #221: Getting local position

vacky11, 2018-07-18:
In sdk version 3.6, how to get local position of the drone using instance of `DJI::OSDK::Vehicle`?
I am looking for a equivalent of the following done with sdk version 3.2
`DJIDrone drone(nh);`
`local_position = drone.local_position`

vacky11, 2018-07-20:
@CWOA Thanks for the explanation. But as mentioned earlier i am looking for a way to access local_position value using instance of `DJI::OSDK::Vehicle`, not by subscribing to a topic. Do you know how can that be done?

CWOA, 2018-07-20:
Im afraid not, I could be wrong but it sounds like you're operating in the onboard sdk, not the onboard **ROS** sdk (this repository), maybe ask your question in that one - https://github.com/dji-sdk/Onboard-SDK

vacky11, 2018-07-20:
ok thanks. I am using Onboard-SDK-ROS

closed 2021-07-07 by dji-kyle