dji-sdk/Onboard-SDK-ROS #222: "flight_control_setpoint_generic" frequency and HORIZONTAL_ANGLE control for PID outer loop position controller

rocklinsuv, 2018-07-20:
Hi @rohitsantdji,

It has been a while since I got stuck with these issues. Currently I have implemented my own PID outer loop controller based on Marvelmind Ultrasonic position feedback. The first issue I encounted is that when I publish to topic "dji_sdk/flight_control_setpoint_generic", it only behaves well when the frequency is no more than 50Hz. If it is greater than 50Hz, say when I let the drone to spin at a certain angular velocity, it spins discontinuously. So my first question will be, is it possible to increase the control publisher rate? What is actually happening when "dji_sdk/flight_control_setpoint_generic" is receiving control messages arriving faster than 50Hz?

The second issue is that I could never get a set of good PID parameters when implementing HORIZONTAL_ANGLE control using both the DJI Assistant 2 simulator's GPS and Marvelmind's position/velocity estimation (Marvelmind outputs are fused with IMU acc using Kalman Filter). I strictly followed the tradicitonal PID tuning guide to first increase Kp to a certain point, then added Ki and Kd terms. However, no matter how small or how larger the Kp values are, it never converged but oscillated in the simulator. Have you guys ever tried implementing a PID outer loop? If you could provide some hints on parameter tuning, it would be very much appreciated.

BR,
rocklinsuv

snakehaihai, 2018-08-13:
even 50hz is too high. for my drone i control it in 40 hz as senser input frequency

rocklinsuv, 2018-08-14:
@snakehaihai Hi there, I found 50Hz working just fine for my drone. Did you try PID outer loop control? If yes, is it possible for you to share your PID parameters? I could never get a good set of P/I/D terms, which is very frustrating. Thanks in advance.

snakehaihai, 2018-08-14:
lol bro. what u ask doesn`t make any sense. since our drone model are different and OSDK version are different oso. mine working on is in 3.2    pith and roll PID are 6/1.5/7.2. But again. its case by case.

koyahatataku, 2019-01-02:
Hi guys, I also tried to use HORIZONTAL_ANGLE with M600Pro but  UAV's angle didn't change at all.

At least you two seem to success to change UAV pitch.

My code is like below:

elapsedTimeInMs = 0;
timeoutInMilSec = 700;
controlFreqInHz = 50; // Hz
while (elapsedTimeInMs < timeoutInMilSec)
{
    ctrl_flag = ( Control::VERTICAL_VELOCITY | Control::HORIZONTAL_ANGLE | Control::YAW_RATE | Control::HORIZONTAL_BODY | Control::STABLE_ENABLE);

    Control::CtrlData data(ctrl_flag, 0, 10, 0, 0);
    vehicle->control->flightCtrl(data);

    usleep(20 * 1000); // 20msec sleep

    elapsedTimeInMs += 20;
} 

Can you guys indicate where of this code is bad?
when I set this code's vertical velocity to some value ,UAV moves vertically, but doesn't move horizontally.

In addition,  I have one suggestion to rocklinsuv.
I think you are trying to calculate UAV angle directly by PID calculation, but maybe at first you should calculate acceleration by PID calcuration, then convert it to UAV angle.
you can convert UAV's acceleration to angle by this formula:

#define RAD2DEG  ((double)180.0 / 3.14159265359)
#define GRAV_ACCEL 9.80665 
angle(degree) = std::atan2( accel, GRAV_ACCEL ) * RAD2DEG;

koyahatataku, 2019-01-03:
Above code works in simulator but doesn't in real world

snakehaihai, 2019-01-07:
20msec... emm i only manage to control my drone at 10 hz.   50 might be too much for serial.  Also which version ur are running and are you running with OSDK alone?   For OSDK-ROS there is multiple version of input code formats.   Currently, the one I'm using is 

   ctrlPosYawPub = nh.advertise<sensor_msgs::Joy>("/dji_sdk/flight_control_setpoint_ENUposition_yaw", 10);
.....

Joy_output_msg.axes.push_back(Position_command_out.linear.y);
 Joy_output_msg.axes.push_back(Position_command_out.linear.x);
Joy_output_msg.axes.push_back(Position_command_out.linear.z);
 Joy_output_msg.axes.push_back(Position_command_out.angular.z);
Joy_output_msg.axes.push_back(flag);
 ctrlVelYawRatePub.publish(Joy_output_msg);   //nonlinear output control

koyahatataku, 2019-01-16:
Thank you for the answer. Now I'm using OSDK3.6 and I'm not using ROS. The command sending frequency(50Hz) is same as OSDK3.6 flight control sample, though it uses HORIZONTAL_VELOCITY flag instead of HORIZONTAL_ANGLE flag. When I once used HORIZONTAL_VELOCITY with 50hz , it worked well.
Anyway lowering the frequency with HORIZONTAL_ANGLE is worth to test(though I cannot test right now).

koyahatataku, 2019-01-16:
[image]
My sdk settings is like above. Do your settings have any differences?

snakehaihai, 2019-01-18:
@koyahatataku I only tested in the real world. I tested before using 40hz from 40-Hz LIDAR based navigation. the result became unstable. Once I lower it to fix 20 Hz. it became stable. That's for LIDAR.
For vision, I lower the Hz rate to 10 Hz.  

Result can be seen in 
https://www.youtube.com/watch?v=1AbfRENy3OQ&feature=youtu.be

u sure you are given a correct range of command for the angle? mine actual control angle is between -0.1 to 0.1

koyahatataku, 2019-01-28:
Thank you for the answer. It seems controlling vertical velocity and horizontal angle at the same time using VERTICAL_VELOCITY and HORIZONTAL_ANGLE flag causes the error.
This works well in simulator (UAV goes down , and  at same the time , moves horizontally), but in real world it seems not to work.
When I set 0 to vertical velocity, UAV seems to   move horizontally.
exactly speaking, I confirm action with listening the motors sounds without propeller, because I cannot fly UAV now(First,I make drone  take off manually)

closed 2021-07-07 by dji-kyle