dji-sdk/Onboard-SDK-ROS #226: Drone takes off and do not move when controlled from a Remote computer via WiFi

sarathjayanthan, 2018-08-07:
Hello,

I am using the Onboard SDK ROS in Raspberry Pi 3 (Ubuntu Mate 16 and ROS Kinetic). I am using the simulation of M600 DJI drone (A3) with DJI Assistant 2.

I was able to successfully fly the drone (only in simulation), by following this link : http://wiki.ros.org/dji_sdk/Tutorials/Getting%20Started . In this case, I had the Raspberry Pi 3 connected to the **wired LAN** and used **Putty for SSH connection**.

But when I connected, Raspberry Pi 3 with my laptop (Ubuntu 16) through **SSH via Wi-Fi connection**, I was not able to fly the drone (again, in simulation only). When I executed **$ roslaunch dji_sdk sdk.launch** and **$ rosrun dji_sdk_demo demo_flight_control**, the drone took off (to 1.2 m) and stayed in the air. The programs were running in the terminal without error, but drone remained in air without moving.

Could anyone help me with this regard? What am I missing here?

Thanks in advance.

snakehaihai, 2019-01-18:
Emm i didn`t encounter this problem.  I control the drone using small wifi and use UVC remote to see it. I`m able to send goal point fron both Ground PC and onboard PC.

camilojimo, 2019-05-27:
Hi I'm running dji ROS sample and I dindt encounter this error. @snakehaihai I'm trying to remap X Y Z with an external controller publishing to /dji_sdk/flight_control_setpoint_generic with this 
flag = (DJISDK::VERTICAL_VELOCITY |
DJISDK::HORIZONTAL_ANGLE |
DJISDK::YAW_RATE |
DJISDK::HORIZONTAL_GROUND |
DJISDK::STABLE_ENABLE)
it works okay on simulation but when I tested on the field, the brake gan looks like it's on 100% have you got any options to reduce this gain to make the drone brake smoothly ?

snakehaihai, 2019-05-31:
I`m not sure about your implementation. For my case, i have the whole solution set up to achieve the soft brake

Camera-LIDAR-Inertial ->  EKF based Locolization 
Camera-LIDAR-Inertial ->  Local+global mapping

locolization+ mapping -> path planning + motion control node

 path planning gives direction and magnitude with respect to UAV base frame. As the drone approaches to the goal point, the motion vector magnitude reduced.

closed 2021-07-07 by dji-kyle