dji-sdk/Onboard-SDK-ROS #228: Reading PWM signals from MFIO

vacky11, 2018-08-16:
Hello,
How to read input PWM signals coming from MFIO pins (F1-F8) of flight controller?

rohitsantdji, 2018-08-17:
Reading PWM that is generated from an external source is not supported through the MFIO pins in the flight controller.

vacky11, 2018-08-17:
@rohitsantdji 
I am not trying to read PWM generated from an external source.
I want to read the shutter signals on ROS, sent by a third-pary waypoint navigation app (supported for dji drones). 
Since I didn't find a way on OSDK-ROS to read these shutter signals, I was planning to direct these shutter signals on one of the function channels of A3 controller(for example F4) and then direct it back into another function channel (for example F5) to read this signal in ROS using pwmLoopback ( similar to gpioLoopbackBlockingApiDemo() function in mfio examples)

Is there a easier way to get the shutter signals in form of a ros topic or something similar?

snakehaihai, 2019-01-18:
@vacky11  you can use an arduino to ream PWM, using rosserial.  THen this pwn is in rostopic system.
Then based on this topic, forward to MIFO sample and start control

closed 2021-07-07 by dji-kyle