dji-sdk/Onboard-SDK-ROS #229: Not able to build using catkin_make for ROS 3.7

imkhairulikhwan, 2018-08-21:
Hi and good day, 

I recently updated the latest Onboard SDK and Onboard SDK ROS 3.7 using following steps: 

https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#ros-onboard-computer

However, I received error message when building using catkin_make:

[image]

dangviethieu-hntv, 2018-08-22:
@imkhairulikhwan you also need update Onboard-SDK v3.7 in https://github.com/dji-sdk/Onboard-SDK
guideline: https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#ros-onboard-computer

imkhairulikhwan, 2018-08-22:
Hi @dangviethieu-hntv ,

thanks for the reply. However, I did update both Onboard-SDK and Onboard-SDK-ROS to v3.7 when I received the above error message.

dangviethieu-hntv, 2018-08-23:
@imkhairulikhwan please try remove lib installed.
_sudo rm -r /usr/local/include/djiosdk
sudo rm -r /usr/local/lib/libdjiosdk-core.a_
And install Onboard-SDK v3.7 again.

imkhairulikhwan, 2018-08-23:
Hi @dangviethieu-hntv , thanks for the prompt reply. Just a quick one, does that mean we couldn't run 3.6 and 3.7 in the same machine, at the same time? (without installing /uninstalling)

dangviethieu-hntv, 2018-08-23:
@imkhairulikhwan Oh, that 's got be it.

imkhairulikhwan, 2018-08-23:
I'm able to finish at 100% now when doing catkin_make. 

Thanks! @dangviethieu-hntv

closed 2018-08-23 by imkhairulikhwan