dji-sdk/Onboard-SDK-ROS #234: No camera images to ROS topics with M210

eeroniemi, 2018-09-04:
After launching ROS master node, I would expect images from different camers to appear on image topics but this doesn't happen. Onboard SDK code camera samples work fine. 

I have been able to reproduce this problem with two computers:

- Linux ARM64 based mini computer (Rock64)
- Intel x64 PC

Both have latest Onboard SDK's installed:

- Onboard-SDK 3.7, commit ID d43cf0ab135b015145ab7a2e41b85cbee6522f83
- Onboard-SDK-ROS 3.7, commit ID 439a8ad

Both computers have Ubuntu 16 + ROS Kinetic. Onboard-SDK is complied with advanced sensing enabled.

## How to reproduce

Connect USB and USB-TTL from M210 to your computer.

Test that camera connection works with Onboard-SDK samples:

```
rock64@rock64:~/dji/Onboard-SDK/build/bin$ ./camera-stream-callback-sample UserConfig.txt
```

Now I can see video stream from both FPV and main camera on my desktop screen.

Launch the DJI ROS node.

```
rock64@rock64:~$ roslaunch dji_sdk sdk.launch
... logging to /home/rock64/.ros/log/f5eb751a-b030-11e8-9738-86e0c0eafaa9/roslaunch-rock64-4257.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rock64:43190/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1060581
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: 4844426e51fd869cc...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [4267]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f5eb751a-b030-11e8-9738-86e0c0eafaa9
process[rosout-1]: started with pid [4280]
started core service [/rosout]
process[dji_sdk-2]: started with pid [4293]
[ INFO] [1536058500.039419784]: Advanced Sensing is Enabled on M210.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 095XF7C002024S

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = PM410

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.10.4

STATUS/1 @ removePackage, L468: Remove package 1 successful.
ERRORLOG/1 @ removeLeftOverPackages, L499: failed to remove package 1
STATUS/1 @ removePackage, L468: Remove package 2 successful.
ERRORLOG/1 @ removeLeftOverPackages, L499: failed to remove package 2
ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ init, L49: Looking for USB device...

STATUS/1 @ init, L65: Found 11 USB devices, identifying DJI device...

STATUS/1 @ init, L83: Found a DJI device...

STATUS/1 @ init, L96: Attempting to open DJI USB device...

STATUS/1 @ init, L134: ...DJI USB device started successfully.

STATUS/1 @ activate, L1277: version 0x3030A04

STATUS/1 @ activate, L1290: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L869: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.[ INFO] [1536058506.819959645]: drone activated
[ INFO] [1536058507.695332577]: Use data subscription to get telemetry data!
[ INFO] [1536058507.695865732]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 2 result: 0.
STATUS/1 @ startPackage, L352: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L350: Start package 1 result: 0.
STATUS/1 @ startPackage, L352: Package 1 info: freq=50, nTopics=14.

STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=5, nTopics=6.

STATUS/1 @ startPackage, L350: Start package 3 result: 0.
STATUS/1 @ startPackage, L352: Package 3 info: freq=400, nTopics=1.
```

List of ROS topics:

```
rock64@rock64:~$ rostopic list
/dji_sdk/acceleration_ground_fused
/dji_sdk/angular_velocity_fused
/dji_sdk/attitude
/dji_sdk/battery_state
/dji_sdk/display_mode
/dji_sdk/flight_anomaly
/dji_sdk/flight_control_setpoint_ENUposition_yaw
/dji_sdk/flight_control_setpoint_ENUvelocity_yawrate
/dji_sdk/flight_control_setpoint_generic
/dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition
/dji_sdk/flight_status
/dji_sdk/fpv_camera_images
/dji_sdk/from_mobile_data
/dji_sdk/gimbal_angle
/dji_sdk/gimbal_angle_cmd
/dji_sdk/gimbal_speed_cmd
/dji_sdk/gps_health
/dji_sdk/gps_position
/dji_sdk/height_above_takeoff
/dji_sdk/imu
/dji_sdk/local_frame_ref
/dji_sdk/local_position
/dji_sdk/main_camera_images
/dji_sdk/rc
/dji_sdk/rc_connection_status
/dji_sdk/stereo_240p_down_back_images
/dji_sdk/stereo_240p_down_front_images
/dji_sdk/stereo_240p_front_depth_images
/dji_sdk/stereo_240p_front_left_images
/dji_sdk/stereo_240p_front_right_images
/dji_sdk/stereo_vga_front_left_images
/dji_sdk/stereo_vga_front_right_images
/dji_sdk/trigger_time
/dji_sdk/velocity
/dji_sdk/vo_position
/rosout
/rosout_agg
```

Now when I try to output images from image topics I get no data. Tried to output images different ways:
 - echoing the image topic `rostopic echo /dji_sdk/fpv_camera_images`
 - by running ROS web video server: `rosrun web_video_server web_video_server` and opening images on browser.
 - via RVIZ

With DJI M100, Onboard-SDK 3.6 and M100 X3 ROS driver (https://github.com/groundmelon/m100_x3) I can see the video feed on topic `/m100_x3/image_raw`

rcabg, 2018-09-04:
When using ROS, did you call the "/dji_sdk/setup_camera_stream" service? It is necessary in order to get the image topics in ROS. See [here](http://wiki.ros.org/dji_sdk#Services) for parameters needed by the service call.

For the main camera, remember to have your RC on with the DJI GO app.

eeroniemi, 2018-09-04:
Seems like I missed that one. After doing that I got images to ROS topics. Thanks for help @rcabg !

closed 2018-09-04 by eeroniemi