dji-sdk/Onboard-SDK-ROS #236: RTK topics not published after upgrading to version 3.7

PaulKemppi, 2018-09-28:
Topics /dji_sdk/rtk_position etc. are not published because in dji_sdk_node.cpp the command 

vehicle->subscribe->initPackageFromTopicList(PACKAGE_ID_5HZ, nTopicRTKSupport,
                                                   topicRTKSupport, 1, 10)

return false. The data transmission rate is set to 10Hz via DJI Assistant 2. All the other topics are published as expected.

droneboxid, 2018-09-30:
Hi Paul-same case with me. i change back to 3.3 version.

aybakana, 2018-10-18:
Hi, I tried version 3.7 for a lot of time and had the same problem. I wish I saw this thread before. This issue should be stated in the Readme though.

aybakana, 2018-10-19:
I contacted delelopers team and They told me that there is a bug in version 3.7. I am using version 3.6 now but still I can not see the RTK topics.  should D-RTK lights turn green to see RTK topics?

PaulKemppi, 2018-11-02:
By downgrading to the version 3.6, one loses the new topics (e.g. visual odometry). Could this bug be corrected ASAP as I need the RTK position for my customer? We bought M210 RTK mainly because of the accurate position it provides, so this bug is really causing trouble to our development.

aybakana, 2019-03-12:
In order to install a previous version of onboard sdk, do you I have to remove the installed one?

closed 2021-07-08 by dji-kyle