dji-sdk/Onboard-SDK-ROS #237: How can I know flight mission is finished?

Seonhoon-Lee, 2018-10-08:
Hello,

I want to generate new mission automatically after previous mission finished, but I can't find a variable which informs mission is finished. Is there any method for knowing whether a mission is finished or not?

aRealBanana, 2019-09-09:
For Android sdk its done like this
`
private WaypointMissionOperatorListener eventNotificationListener = new WaypointMissionOperatorListener() {
        @Override
        public void onDownloadUpdate(WaypointMissionDownloadEvent downloadEvent) {

        }

        @Override
        public void onUploadUpdate(WaypointMissionUploadEvent uploadEvent) {

        }

        @Override
        public void onExecutionUpdate(WaypointMissionExecutionEvent executionEvent) {

        }

        @Override
        public void onExecutionStart() {

        }

        @Override
        public void onExecutionFinish(@Nullable final DJIError error) {
            setResultToToast("Execution finished: " + (error == null ? "Success!" : error.getDescription()));
            if(error==null)
           {
               
            }
        }
    };
`

dji-kyle, 2021-07-08:
mission callback:
setWaypointEventCallback
setWaypointCallback

closed 2021-07-08 by dji-kyle