dji-sdk/Onboard-SDK-ROS #238: ROS support DJI spark

rfsch, 2018-10-15:
I am trying to control a DJI spark via WIFI from a computer using ROS. Could you please let me know whether the Spark supports control via ROS. 

Please suggest the best way to control flight path from a remote computer programmatically. 
Our task is to scan a room interior using a special TOF camera.
We would like to control the spark from a remote computer (running ROS) based on the quality of the image (point cloud) captured by our special TOF camera. That is we would like to tell the spark to fly where the quality of the captured image is bad so we can automatically re-scan until satisfied.

chutsu, 2018-11-04:
Probably not, they ignored me back when it first came out #90.

rfsch, 2018-11-04:
thanks. It seems the new Windows SDK does not support the Spark as well :(

chutsu, 2018-11-04:
If you don't mind the data lag have you considered the Parrot Bebop? That drone has a ROS SDK and you can control it via wifi etc.

rfsch, 2018-11-04:
Thanks for the info. But unfortunately I already have a Spark for a project where I want to make room scans. Got a Raspberry PI plus a TOF camera attached to the Spark which sends the data via ROS to a Windows machine. Now I need a way to control the Spark based on the quality of the scans that were captured. Guess I am stuck with getting the Mobile SDK work inside a VM on Windows...  Or do you have some better idea ?

chutsu, 2018-11-04:
Afraid not :S 
Good luck!

closed 2021-07-08 by dji-kyle