dji-sdk/Onboard-SDK-ROS #241: Build Errors cmake..

BrMerrell, 2018-11-08:
I get the following error when doing the documented install. Is this a known issue or just me? 
~/Onboard-SDK-ROS/dji_sdk/build$ cmake ..

CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "FindDJIOSDK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DJIOSDK", but
  CMake did not find one.

  Could not find a package configuration file provided by "DJIOSDK" with any
  of the following names:

    DJIOSDKConfig.cmake
    djiosdk-config.cmake

  Add the installation prefix of "DJIOSDK" to CMAKE_PREFIX_PATH or set
  "DJIOSDK_DIR" to a directory containing one of the above files.  If
  "DJIOSDK" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

closed 2018-11-08 by BrMerrell

vahidbehtajisiahkal, 2019-07-25:
Hi
I also got the same error anyone can help me?

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "dji_sdk" with any
  of the following names:

    dji_sdkConfig.cmake
    dji_sdk-config.cmake

pavankumarbn, 2019-07-25:
Hi,

This thread might help you 
https://answers.ros.org/question/244843/cmake-cannot-find-package-for-dji-onboard-sdk-file/

BW,
Pavan