dji-sdk/Onboard-SDK-ROS #244: [Version 3.7][M210 RTK] No work setWaypointEventCallback

btsatoukzn, 2018-12-03:
I am using osdk-ros 3.7. And we added a method to check if we arrived at waypoint during mission execution. However, when arriving at a waypoint, that method will not be executed.

aircraft firmware is V01.02.0000.

Please tell me this solution.

Added to missionWpUploadCallback function of dji_sdk_node_mission_services.cpp.

```
vehicle->missionManager->wpMission->setWaypointEventCallback(waypointEventCallback, 
(DJI::OSDK::UserData) this);
```

Add waypointEventCallback.

```void
DJISDKNode::waypointEventCallback(Vehicle* vehicle, RecvContainer recvFrame, DJI::OSDK::UserData userData)
{
  DJISDKNode *p = (DJISDKNode *)userData;
  dji_sdk::MissionIndex mission_index;
  mission_index.waypoint_index = recvFrame.recvData.wayPointReachedData.waypoint_index;
  mission_index.current_status = recvFrame.recvData.wayPointReachedData.current_status;
  mission_index.incident_type = recvFrame.recvData.wayPointReachedData.incident_type;
  p->mission_index_publisher.publish(mission_index);
}
```

btsatoukzn, 2018-12-10:
I tried pull request's solution. but no work too...
[https://github.com/dji-sdk/Onboard-SDK-ROS/pull/186](https://github.com/dji-sdk/Onboard-SDK-ROS/pull/186)

btsatoukzn, 2018-12-17:
Last week, we went outdoor flying M210. At that time, setWaypointEventCallback worked.

Until now, it did not work on the simulator.

Is that something like that?

aanast01, 2019-09-06:
Hello @btsatoukzn. Did you find any solution to the problem? I've tried lots of things but with no luck. The callback is never called neither on the simulator nor in a real flight.

realuptime, 2020-06-16:
For those of you that has this issue, I amanaged to solve it by linking to these static libraries:

    OSDK_LIBS=~/Onboard-SDK/libs/
    LDFLAGS+=-rdynamic $(OSDK_LIBS)/libadvanced-sensing.a $(OSDK_LIBS)/libdjiosdk-core.a $(OSDK_LIBS)/libadvanced-sensing.a $(OSDK_LIBS)/../osdk-core/linker/x86/lib/libdji-linker.a

dji-kyle, 2021-07-08:
Please check the DJI Assistant2 OSDK interface and enable "Ground Station Status"
[image]

closed 2021-07-08 by dji-kyle