dji-sdk/Onboard-SDK-ROS #247: Drone does not fly further than 20 meters

Gerjen100, 2018-12-13:
Hi, 

I have been working on a program that uses the dji_sdk/flight_control_setpoint_generic as a way of controlling the drone. This works fine on the when we use the M100. However when we use M600 the drone abruptly stops when it is 20 meters away from me.

This however only happens when we are in a certain location. We fly at this location because it is a certified testing area. It used to be a restricted area but it got unlocked about a year ago. We can fly manually but when we use the osdk to control the drone this issue occurs.

Im not sure if this is the right place to ask for help seeing the unique condition of our problem, but if not i would like to know if anyone could refer me to someone could help us resolve this issue.

Kind regards Gerjen

snakehaihai, 2019-01-18:
emm. I didn`t experience this. I wrap all DJI GPS and compass with aluminum foils.  My drone is controlled by vision-based navigation and GPS based navigation. Can fly without any limit.

phwoo, 2019-04-11:
@Gerjen100 Did you end up resolving this issue?

lanyusea, 2019-04-11:
It may be limited when drone in vision-based navigation.

should not be limited when the drone can in GPS mode. I suggest to check the GPS signal strength

snakehaihai, 2019-04-16:
The vision drone I was using is a small stereo based drone using N3. Whether it is on special firmware i dont know, as there are other people help me deal with controller level thing. I only control high-level algorithm.  the video is here.  https://www.youtube.com/watch?v=9NuLg3ouWmM  

If you are using internal GPS based routine then you probably need to get premium DJI membership / DJI partnership, Then they will give you special firmware to do whatever u want ( for example partnership can give you ability to remove GPS limit. can allow drone to arm with EMI. early MSDK access, full PSDK access, early flyhub access, arm the drone in the airport or military installation on moments notice etc)

rikba, 2019-06-03:
I think it's a safety thing. You need to attach your cellphone with DJI GO app to the remote. Then you should be able to fly further under SDK control.

Gerjen100, 2019-06-04:
Yeah i resolved the issue in the past by connecting a mobile device to the controller on other locations. But i think it has to do with the fact that the location is restricted. We do have the code to fly the drone there however the SDK still somehow restricts the flight to no further than 20 meters .

closed 2021-07-08 by dji-kyle