dji-sdk/Onboard-SDK-ROS #248: is Matrice drone a reliable platform for autonomous indoor mission ?

MI-LA01, 2019-01-02:
Sounds challenging to implement any external positioning system (UWB, Optitrack etc. ). In respect of this, I notice also the poor (or lack) of answers from Dji dev-team in the repo and almost nothing on stackoverflow. 

Consequently, could You kindly share Your mind if Matrice (100 - 2x0) platforms are reliable for autonomous indoor flights ? Rgds a

andre-nguyen, 2019-01-03:
@MI-LA01 It all depends on you and what degree of competency you have in gluing together existing ros packages and writing your own controllers/estimators. Personally, I have used an m100 indoors with a vicon system when I was a student @mrasl and there are multiple other labs who use them indoors e.g. @jhu-asco and @ethz-asl . Some commercial companies also use the m100 platform for gps denied navigation including @automodality and Exyn Technologies.

The reality however is that you'll need to at least put some work in, even if you get the m100 with the guidance module, you might need to write your own position controller or fuse the guidance odometry data with your own global localization system. A lot of people were able to do indoor navigation with the Ascending Technologies research line and those drones had less features than the DJI ones.

MI-LA01, 2019-01-03:
Hello @andre-nguyen and thank You so much for sharing this interesting feedback. In respect of UWB or external positioning system (i.e : position controllers) , would You kindly share some literatures, references or c++ samples that could benefit for the community and people reading this thread ?

I spent time to read the official API from Dji website and samples with no-luck. I'm a bit afraid about the lack of tutorials and documentation for advanced usage that would require high time consuming for prototyping something on this platform. 

Best rgds.

andre-nguyen, 2019-01-03:
Maybe you can have a look here https://github.com/ethz-asl/mav_dji_ros_interface/wiki however their repo was forked before the onboard SDK got a big rewrite. Where they use Rovio you could plug your uwb or vicon system to MSF instead.

reygago, 2019-01-15:
Hi @MI-LA01 !
Since 2017, I've been working with Matrice 100 with self localization. It's been a long path 'til here... :)
I've also researched some solutions based on UWB and external cameras (not priority for my job), the main thing about any of these solutions is that you have to keep track of your local position with your own hardware/system, **you cannot update the drone inner systems with your own localization system**. Usually, you just send attitude commands for positioning. If your solution allows this kind of workaround, it will be fine.
About Matrice 100, the platform is very stable and robust. I'm flying with 3.8 kg, a little bit more than recommeded and, despite the low flight time, usually the drone has good response. Also, I've crashed more than I wished for during tests, and with some spare parts I was able to rebuild the drone most of times, with 1-day work.

Look for papers like "Dynamic System Identification, and Control for a cost effective
open-source VTOL MAV" and "Indoor Navigation of Quadrotors via Ultra-Wideband Wireless Technology", they can give you some insight.

I hope that I will publish something this year, you can see a video of the solution I'm building in http://bit.ly/2zy4jdR

closed 2021-07-08 by dji-kyle