dji-sdk/Onboard-SDK-ROS #249: How to hover in waypoint mission?

zhudaoruyi, 2019-01-03:
We know that we can set actions in a specific mission waypoint, but
 how to hover?

snakehaihai, 2019-01-18:
@zhudaoruyi   just set goal point at while loop to be the point u want to stay. each loop with 1-sec delay. EZ

@dji-sdk  useless official channel. I reported a bug of MIFO. Takes half year to confirm the bug. still far  away from a solution. By the time official support come with a solution, the project is long over.   Here is better because community might help

dji-kyle, 2021-07-08:
Comment on OSDK 4.x version, same as OSDK 3.9
/**
 * @brief Waypoint settings for individual waypoints being added to the mission
 * @details This is one of the few structs in the OSDK codebase that
 * is used in both a sending and a receiving API.
 */
typedef struct WayPointSettings
{
  uint8_t   index;     /*!< Index to be uploaded */
  float64_t latitude;  /*!< Latitude (radian) */
  float64_t longitude; /*!< Longitude (radian) */
  float32_t altitude;  /*!< Altitude (relative altitude from takeoff point) */
  float32_t damping; /*!< Bend length (effective coordinated turn mode only) */
  int16_t   yaw;     /*!< Yaw (degree) */
  int16_t   gimbalPitch; /*!< Gimbal pitch */
  uint8_t   turnMode;    /*!< Turn mode <br> */
  /*!< 0: clockwise <br>*/
  /*!< 1: counter-clockwise <br>*/
  uint8_t reserved[8]; /*!< Reserved */
  uint8_t hasAction;   /*!< Action flag <br>*/
  /*!< 0: no action <br>*/
  /*!< 1: has action <br>*/
  uint16_t actionTimeLimit;      /*!< Action time limit */
  uint8_t  actionNumber : 4;     /*!< Total number of actions */
  uint8_t  actionRepeat : 4;     /*!< Total running times */
  uint8_t  commandList[16];      /*!< action list.Take value from enum WaypointActionTypeFormat.*/
  int16_t  commandParameter[16]; /*!< action parameters.Details Please see notes of enum WaypointActionTypeFormat's action*/
} WayPointSettings;              // pack(1)

typedef enum WaypointActionTypeFormat
{
  WP_ACTION_STAY                 = 0,  /*!< no action.uint of action parameter:ms*/
  WP_ACTION_SIMPLE_SHOT          = 1,  /*!< take picture action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_VIDEO_START          = 2,  /*!< start take video action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_VIDEO_STOP           = 3,  /*!< stop video action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_CRAFT_YAW            = 4,  /*!< craft control yaw action.uint of action parameter:degree. range:-180 ~ 180*/
  WP_ACTION_GIMBAL_PITCH         = 5,  /*!< gimbal control pitch action.uint of action parameter:degree. range:-90 ~ 0*/
} WaypointActionTypeFormat;

closed 2021-07-08 by dji-kyle