dji-sdk/Onboard-SDK-ROS #259: GPS pos covariance

sungwook87, 2019-02-14:
Hi,
I have been using A3/A3 pro with M600/custom platform.
The thing is, how can I get the position covariance of GPS?
In /dji_sdk/gps_position topic, it is always '0' in pos cov.
does it basically come from the FC? or Should I set any parameters in SDK?
Thanks in advance!

lanyusea, 2019-02-15:
According to the code: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/72729f13709dcaf2ae94aaf9cb4f26ddaf42f958/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L120-L124

The gps_pos should be come from the gps measurement of FC.

Some possible reason and trial:
1. check the GPS data broadcast in DJI Assistant
2. try fly in a place with better GPS signal
or
use the simulator and set the GPS to a proper value
3.

dunk-emesent, 2019-02-15:
There is no gps covariance supplied by the drone. I'm not sure what the DJI
ROS node does, but it'll have to derive covariance from vdop and hdop -
these values are published by the drone.

sungwook87, 2019-02-15:
@dunk-emesent @lanyusea 

Yeap, exactly. It only gives us the health in integer value.
It seems FC calculates the health with the covariance info. internally.
I guess we need to set the proper covariance intuitionally according to the GPS health value.

lanyusea, 2019-02-15:
@dunk-emesent  @sungwook87 

yes, the health is the result calculated inside by GPS covariance,  SVN, etc. You can treat it as the reliability. 

I suggest 3 (range is from 0 to 5 as I remember) if you need a threshold value.

sungwook87, 2019-02-15:
@lanyusea 
Thanks, mate :)
Closing this issue.

sungwook87, 2019-02-15:
@lanyusea 
By the way, do you remember me? Are you the guy visited our lab. a few years ago?
You must be from the prof. Shaojei Shen's lab. Good to see you here again.

closed 2019-02-15 by sungwook87

lanyusea, 2019-02-15:
@sungwook87  wow yes it was me. glad to meet you here.