dji-sdk/Onboard-SDK-ROS #260: Why dji_sdk_dji2mav project folder has been removed after 3.2 release

Aansh2, 2019-02-15:
dji_sdk_dji2mav project  has been remove after  3.2 version. Can any one suggest alternate project of it or can I still use it to connect DJI drone N3 model to remote GCS?

lanyusea, 2019-02-15:
I think you can still use it.

petergerten, 2019-03-20:
I am also very interested in this. Can someone confirm if it is still working when used with the latest version? Any missing core functionality?

petergerten, 2020-01-02:
Anyone working with this?

closed 2021-07-08 by dji-kyle

petergerten, 2021-07-09:
@dji-kyle it would be much appreciate if you could at least provide **ANY** comment on the issue before closing. :-(

dji-kyle, 2021-07-12:
@petergerten I am sorry, For N3 please update the OSDK version to 3.9 and ROS to 3.8, if there is any issues of the latest version, please create new.

dji-kyle, 2021-07-12:
@petergerten I am sorry, For N3 please update the OSDK version to 3.9 and ROS to 3.8, if there are any issues, please create new. The old version will no longer provide technical support and maintenance.