dji-sdk/Onboard-SDK-ROS #261: Thermal XT2 Parameters access with OSDK

guicolares, 2019-02-15:
Hi everyone, 

We have a project to get Thermal Camera Image with OSDK and get some parameters such as 
palette type, temperature data, etc. 

Is it possible get all this data (from Thermal Camera) direct from OSDK ?  We can do that with Mobile SDK but we need only with Onboard.

We are using a M210 and TX2 (Nvidia). 

Thanks.

gssales, 2019-09-03:
I am having the same problem

Have you found a solution yet?

rodbelisario, 2020-08-12:
Hi Guys, any news about this topic?
I'm also trying to get the thermal frames from XT2 on m210 using OSDK.
Any one had lucky on that?
Thanks in advance

dji-kyle, 2021-07-08:
OSDK only supports to get the video stream, can't get palette type, temperature data.

closed 2021-07-08 by dji-kyle