dji-sdk/Onboard-SDK-ROS #263: DJI Onboard-SDK-ROS : How to change Telemetry Topics Subscription frequency

k-koh, 2019-02-22:
In DJI Onboard-SDK-ROS, sdk push topics

    dji_sdk/attitude
    dji_sdk/angular_velocity_fused

in DJISDKNode::publish100HzData() with 100Hz

and push topics

    dji_sdk/vo_position
    dji_sdk/velocity
    dji_sdk/height_above_takeoff

in DJISDKNode::publish50HzData() with 50Hz.

But i find the result frequency for topics(vo_position, velocity, height_above_takeoff) only have 1Hz in matrice m210 rtk.

How to change Telemetry Topics Subscription frequency to receive thes topic data in high frequency?

lanyusea, 2019-02-25:
It can be adjusted by the DJI Assistant 2 software as I remember.

k-koh, 2019-02-26:
> It can be adjusted by the DJI Assistant 2 software as I remember.

Thank you for your reply.
I set baud_rate in DJI Assistant 2 , but it no affect.
when i set baud_rate = 921600 in sdk.launch, 
 i can not launch ros dji_sdk_node for drone activation error.
i debug activation error to find that dji sdk got empty rawACK in getDroneVersion().
[image]
[image]
[image]

lanyusea, 2019-02-26:
@k-koh not the baudrate, but the following data frequency

(correct me if I misunderstand your question)

closed 2021-07-08 by dji-kyle