dji-sdk/Onboard-SDK-ROS #268: Can not catkin_make Onboard-SDK-ROS

shadow418, 2019-04-12:
I built along this page.

[https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html#linux-oes](url)

But, the following error occurred.
Do you know the solution?

[image]

snakehaihai, 2019-04-16:
hi koki are you sure ur OSDK and OSDK-ROS version are same?
OSDK in ur OS has to be compatible with OSDK-ROS
It is not a compatibility issue. then i`m guessing u are using the latest version. quit using the latest version, 3.6 and 3.7 are more mature.

huangzonghao, 2019-04-29:
The changes introduced in onboard-sdk ver. 3.8 is causing problem here. If you need to run onboard-sdk-ros right away, rolling back to ver. 3.7 could be a temporary fix.

lanyusea, 2019-04-29:
@huangzonghao good catch, I will see if I have time to give a fix

dji-kyle, 2021-07-08:
Please update to OSDK 3.9 + ROS 3.8

closed 2021-07-08 by dji-kyle