dji-sdk/Onboard-SDK-ROS #269: demo_flight_control truble

slamlixiaodong, 2019-04-30:
I ran sdk.launch file and demo_flight_control node, the drone in the simulator can takeoff. However, it stay there always.  demo_flight_control node control station output the same words always as follow.  I set a flag and output it in the flight_control callback function, and through the rostopic info, I can see the topic which can be subscribed by dji_ sdk node, but the callback function did not output ths flag, so I think the callback function did not execute, how can I solve the problem.   

[ INFO] [1556584937.076822715]: +++++dx=-0.000001, dy=20.000000, dz=2.883188, dyaw=0.449960 ...
[ INFO] [1556584937.918653449]: -----x=0.000002, y=-0.000000, z=0.116812, yaw=1.497157 ...
[ INFO] [1556584937.918739559]: +++++dx=-0.000002, dy=20.000000, dz=2.883188, dyaw=0.449960 ...
[ INFO] [1556584938.696716824]: -----x=0.000001, y=0.000000, z=0.116812, yaw=1.497157 ...
[ INFO] [1556584938.696800632]: +++++dx=-0.000001, dy=20.000000, dz=2.883188, dyaw=0.449960 ...
[ INFO] [1556584939.556543578]: -----x=-0.000000, y=-0.000000, z=0.116812, yaw=1.497157 ...
[ INFO] [1556584939.556623833]: +++++dx=0.000000, dy=20.000000, dz=2.883188, dyaw=0.449960 ...
[ INFO] [1556584940.376601964]: -----x=-0.000000, y=-0.000000, z=0.116812, yaw=1.497157 ...
[ INFO] [1556584940.376730475]: +++++dx=0.000000, dy=20.000000, dz=2.883188, dyaw=0.449960 ...
[ INFO] [1556584941.196652578]: -----x=-0.000000, y=-0.000000, z=0.116812, yaw=1.497157 ...
[ INFO] [1556584941.196740328]: +++++dx=0.000000, dy=20.000000, dz=2.883188, dyaw=0.449960 ...

zxy20162101, 2019-05-13:
i met the same problem.

slamlixiaodong, 2019-05-14:
> i met the same problem.

I think you should update the firmware via the Firmware Update tab in the assistant2. If you use a sbus reciever, I advised you that don't update the newest version of fireware.

closed 2021-07-08 by dji-kyle