dji-sdk/Onboard-SDK-ROS #270: Yaw setpoint overshoots

dakejahl, 2019-06-11:
**Airframe**: M600 Pro
**Onboard-SDK version:** 3.7
**Onboard-SDK-ROS version:** 3.6

I want to control ground frame velocity and yaw angle. I have set the flags as such:
```
    uint8_t flag = (DJISDK::VERTICAL_VELOCITY |
                    DJISDK::HORIZONTAL_VELOCITY |
                    DJISDK::YAW_ANGLE |
                    DJISDK::STABLE_ENABLE |
                    DJISDK::HORIZONTAL_GROUND);
```
I am using `dji_sdk/flight_control_setpoint_generic` and publishing at 50Hz.

I am using a PID controller to convert my desired position into velocity setpoints, and this works well. I am sending a yaw setpoint which is a function of the angle between my current heading and my setpoint heading. The problem I am experiencing is large overshoot with yaw.

Can someone please comment on this? It seems like the A3 is not doing a good job at controlling the yaw.

dji-dev, 2019-06-11:
> Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/24838'>Zendesk ticket #24838</a>:<br><br>How do you set the yaw angle? We tested it and it really has overshoot. But the aircraft will quickly return to the set angle.

dakejahl, 2019-06-11:
I am calculating the yaw angle from the desired postion vector `<E, N, U>` 
`yaw = atan (N / E)`

Unfortunately any amount of overshoot is unacceptable for my application. When I command setpoints, I would expect the A3 to be tuned for the M600 for an optimized response.

dakejahl, 2019-06-11:
I should also mention, the overshoot is substantial. When yawing from 0 degrees to 180 degrees, it overshoots by about 45 degrees, and then overshoots going the other way by about 20 degrees, and then finally settles.

dakejahl, 2019-06-17:
Should I contact someone at DJI customer support directly about this?

InterestingWalrus, 2019-06-26:
@dakejahl, could you kindly please point me to the code for your PID control? I'm also trying to do something similar using the same control flag you're using although on the M100 and I'm experiencing overshoot on all the axes

InterestingWalrus, 2019-06-28:
@dakejahl , my issue is an overshoot in the final position setpoints. I expected an improvement on the default control scheme by adding a PID controller but in most cases, it's marginal or sometimes worse compared to not using a PID. I'm doing something similar to yours but I'm using the step function in the examples:

```
void FlightPlanner::stepPID(sensor_msgs::NavSatFix &current_gps, geometry_msgs::Quaternion &current_atti)
{
  geometry_msgs::Vector3     localOffset; 
  static bool finished_x = false;
  static bool finished_y = false;
  static bool finished_z = false;
  static bool finished_yaw = false;

  float yawThresholdInDeg = 170;
  float xCmd, yCmd, zCmd, yawCmd;

  localOffsetFromGpsOffset(localOffset, current_gps, start_gps_location);
  double xOffsetRemaining = target_offset_x - localOffset.x;
  double yOffsetRemaining = target_offset_y - localOffset.y;
  double zOffsetRemaining = target_offset_z - localOffset.z;

  double yawThresholdInRad = Deg_To_Rad(yawThresholdInDeg);
  double yawInRad          = toEulerAngle(current_atti).z;
  double yawDesiredRad = atan2(xOffsetRemaining, yOffsetRemaining);

if(std::abs(xOffsetRemaining) < 0.3 && std::abs(yOffsetRemaining) < 0.3 && std::abs(zOffsetRemaining) < 0.3 )
{
    // We've reached x waypoint
    finished_x = true;
    finished_y = true;
    finished_z = true;
    xCmd = yCmd = zCmd = 0;
}

else
{
    finished_x = false;
    finished_y = false;
    finished_z = false;
    xCmd = pid_x.PID_update(current_local_position.x, target_offset_x, dt);  
    yCmd = pid_y.PID_update(current_local_position.y, target_offset_y, dt);   
    if(waypoint_index == 0)
    {
         zCmd = pid_z.PID_update(current_local_position.z, target_offset_z + takeoff_altitude, dt); 
    }

    else
    {
        zCmd = pid_z.PID_update(current_local_position.z, target_offset_z, dt); 
    }
    
      
    yawCmd = 0;
}

```
I'm assuming the argument to that function is an ENU setpoint? Is there any advantage to vectorization like you have done compared to just setting the velocities on each individual axis?

dakejahl, 2019-06-30:
Looks like you are sending position setpoints? It's hard for me to help with only that snippet of code. Vectorization for velocity control is important to me becauseI want to fly a straight line path between waypoints. This actually fails when you start to control yaw while at the same time controlling ENU velocities. Anyways, it sounds like DJI has not tuned their flight controllers very well for the use case of offboard position setpoints.

closed 2020-01-09 by dakejahl