dji-sdk/Onboard-SDK-ROS #273: time alignment

asherikov, 2019-07-30:
I have doubts regarding the time alignment logic implemented in `alignRosTimeWithFlightController()` (https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L710). The time when a tick message is received is always larger than the time when it is sent (on the same clock), so approximating the actual time offset between the clocks with the minimal observed offset value is better than chosing the offset so that time erros are within dt=TIME_DIFF_CHECK.

dji-dev, 2019-07-31:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26604'>Zendesk ticket #26604</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The question needs further analysis and testing, if need further communication or there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

dji-dev, 2019-08-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26604'>Zendesk ticket #26604</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your feedback and patience. As for your requirements, we will consider adding them into the later support plan. However, as you know, software development needs certain time and uncertainty, so we cannot guarantee the expected release version and time with you. It is recommended that you continue to follow the release and product trends of OSDK, or you can customize the code to implement a part of the requirements.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2020-05-15 by asherikov