dji-sdk/Onboard-SDK-ROS #274: 妙算2和M600在在使用OnboardSdkRos的时候，妙算找不到M600的窜口标识

pzy2253999075, 2019-08-04:
你好，我在M600 PRO 和 妙算2G 上测试Onboard SDK ROS版的时候，发现妙算识别不了M600 PRO的窜口，但是ROS的 launch文件启动需要此窜口标识，我在妙算终端输入 lsusb 和  ls /dev/tty* 都没有发现无人机的窜口（因为没有新窜口增加），请问如何解决呢，谢谢。具体描述可以查看下面几张图片：
[image]
[image]
[image]
[image]

dji-dev, 2019-08-05:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26726'>Zendesk ticket #26726</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

有注意到您使用的使用Manifold2-G，如果使用UART，只能使用UART1连接到M600 Pro，设备名称为：ttyTHS2
建议您使用Manifold2前可以先详细阅读产品使用手册，了解相关必要信息和注意事项：https://dl.djicdn.com/downloads/manifold-2/20190528/Manifold_2_User_Guide_v1.0_CHS.pdf
 [image]
感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-05:
@dji-dev 谢谢，但是我有遇到新的问题
[image]
这是我的M600 PRO 硬件版本以及SDK调试页面
[image]
[image]

请问这个怎么解决呢

dji-dev, 2019-08-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26726'>Zendesk ticket #26726</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

有注意到您使用的是M600 Pro，该机型不支持AdvancedSensing功能，因此需要把ADVANCED_SENSING功能关掉，可以修改dji_sdk_node.cpp的代码，参考操作如下： [image]​
修改保存后，重新编译即可

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/mLMb2YwVBxKO9VIkpV7a7yMMg' target='zd_attach'>inline-1122641980.png</a>

pzy2253999075, 2019-08-06:
@dji-dev 嗯 可以运行了，但现在我需要通过can窜口接妙算和无人机（uart0和1用于接其他设备），比如我接can0，那对应的设备窜口是哪个呢，再次感谢
[image]

dji-dev, 2019-08-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26726'>Zendesk ticket #26726</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

需要注意的是 OSDK API 接口不支持CAN总线，建议您可以使用USB-TTL模块。
另外CAN设备在Linux下是网络设备，can0就是对应的设备名称。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-06:
@dji-dev 好的，谢谢，那M600的can接口和妙算的can接口相连也不支持是吧？

dji-dev, 2019-08-07:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26726'>Zendesk ticket #26726</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

是的，M600的CAN接口暂不支持二次开发。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

JoenHune, 2021-06-13:
> Public comment from Bruce Cheung in [Zendesk ticket #26726](https://djisdksupport.zendesk.com/agent/tickets/26726):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 有注意到您使用的使用Manifold2-G，如果使用UART，只能使用UART1连接到M600 Pro，设备名称为：ttyTHS2
> 建议您使用Manifold2前可以先详细阅读产品使用手册，了解相关必要信息和注意事项：https://dl.djicdn.com/downloads/manifold-2/20190528/Manifold_2_User_Guide_v1.0_CHS.pdf
> [image]
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持

您好，那么请问CPU版本的妙算2，它的uart0口有办法调整为普通串口使用吗？外接usb2tty设备会太复杂了

dji-kyle, 2021-07-08:
妙算2C uart0是预留用于debug的串口，固定波特率为115200，无法直接用于OSDK通信使用。

closed 2021-07-08 by dji-kyle