dji-sdk/Onboard-SDK-ROS #276: osdk-3.8 launch error

chaitanyantr, 2019-08-06:
drone - m600 pro 

Build the osdk-ros getting this error tried restart ,new app id , log files attached below. 

started roslaunch server http://tegra-ubuntu:35539/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1074920
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: b20e1730e4e067fdf...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [30015]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9236ca4c-b836-11e9-9bfe-00044bc4fd3e
process[rosout-1]: started with pid [30041]
started core service [/rosout]
process[dji_sdk-2]: started with pid [30058]
[ INFO] [1565088015.697559872]: Advanced Sensing is Enabled on M210.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0670155018

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14

STATUS/1 @ removePackage, L468: Remove package 0 successful.
ERRORLOG/1 @ removeLeftOverPackages, L499: failed to remove package 0
ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

ERRORLOG/1 @ AdvancedSensing, L58: Please make sure the connected drone is a M210 and firmware version is supported.

STATUS/1 @ activate, L1313: version 0x302290E

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.
[dji_sdk-2] process has died [pid 30058, exit code -11, cmd /home/nvidia/catkin_dji/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/nvidia/.ros/log/9236ca4c-b836-11e9-9bfe-00044bc4fd3e/dji_sdk-2.log].
log file: /home/nvidia/.ros/log/9236ca4c-b836-11e9-9bfe-00044bc4fd3e/dji_sdk-2*.log

[master.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3471750/master.log)
[roslaunch-tegra-ubuntu-29979.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3471751/roslaunch-tegra-ubuntu-29979.log)
[rosout.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3471752/rosout.log)
[rosout-1-stdout.log](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3471753/rosout-1-stdout.log)

chaitanyantr, 2019-08-06:
**Solved**

[image]

closed 2019-08-06 by chaitanyantr

fanshi14, 2020-01-11:
@chaitanyantr 
Thanks, it really help.
___________
For M100 users, please just remove all the ADVANCED_SENSING related codes, and it will work.

JavascriptAddict, 2020-02-10:
@fanshi14 Thanks for the help mate! We were wondering what was wrong with the SDK...