dji-sdk/Onboard-SDK-ROS #277: Onboard_sdk_3.7_Build_Error -

chaitanyantr, 2019-08-06:
**Got the ROS OSDK from github followed the instruction in dev.dji ,getting the build problem as follows**  

tried to run the latest osdk 3.8 but got this [issue](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/276) , so i tried to run the 3.7 got this build problem.

 (i don't want the demo missions services)

[ 89%] Built target dji_sdk_generate_messages_eus
[ 90%] Built target demo_flight_control
[ 90%] Built target demo_camera_gimbal
[ 91%] Built target demo_mfio
[ 92%] Built target demo_local_position_control
[ 93%] Built target demo_mobile_comm
[ 94%] Building CXX object Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_mission_services.cpp.o
[ 94%] Built target dji_sdk_generate_messages
[ 95%] Built target demo_mission
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionStatusCallback(dji_sdk::MissionStatus::Request&, dji_sdk::MissionStatus::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:20:62: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘wayptCounter’
   response.waypoint_mission_count = vehicle->missionManager->wayptCounter;
                                                              ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:21:62: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   response.hotpoint_mission_count = vehicle->missionManager->hotptCounter;
                                                              ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionWpActionCallback(dji_sdk::MissionWpAction::Request&, dji_sdk::MissionWpAction::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:131:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘wayptCounter’
   if (vehicle->missionManager->wayptCounter == 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionWpGetSpeedCallback(dji_sdk::MissionWpGetSpeed::Request&, dji_sdk::MissionWpGetSpeed::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:188:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘wayptCounter’
   if (vehicle->missionManager->wayptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionWpSetSpeedCallback(dji_sdk::MissionWpSetSpeed::Request&, dji_sdk::MissionWpSetSpeed::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:215:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘wayptCounter’
   if (vehicle->missionManager->wayptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionWpGetInfoCallback(dji_sdk::MissionWpGetInfo::Request&, dji_sdk::MissionWpGetInfo::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:247:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘wayptCounter’
   if (vehicle->missionManager->wayptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionHpActionCallback(dji_sdk::MissionHpAction::Request&, dji_sdk::MissionHpAction::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:325:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   if (vehicle->missionManager->hotptCounter == 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionHpGetInfoCallback(dji_sdk::MissionHpGetInfo::Request&, dji_sdk::MissionHpGetInfo::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:383:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   if (vehicle->missionManager->hotptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionHpUpdateYawRateCallback(dji_sdk::MissionHpUpdateYawRate::Request&, dji_sdk::MissionHpUpdateYawRate::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:416:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   if (vehicle->missionManager->hotptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionHpResetYawCallback(dji_sdk::MissionHpResetYaw::Request&, dji_sdk::MissionHpResetYaw::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:454:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   if (vehicle->missionManager->hotptCounter > 0)
                                ^
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp: In member function ‘bool DJISDKNode::missionHpUpdateRadiusCallback(dji_sdk::MissionHpUpdateRadius::Request&, dji_sdk::MissionHpUpdateRadius::Response&)’:
/home/nvidia/ck_dji/src/Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp:491:32: error: ‘class DJI::OSDK::MissionManager’ has no member named ‘hotptCounter’
   if (vehicle->missionManager->hotptCounter > 0)
                                ^
Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/build.make:158: recipe for target 'Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_mission_services.cpp.o' failed
make[2]: *** [Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/src/modules/dji_sdk_node_mission_services.cpp.o] Error 1
CMakeFiles/Makefile2:2256: recipe for target 'Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/all' failed
make[1]: *** [Onboard-SDK-ROS/dji_sdk/CMakeFiles/dji_sdk_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

dji-dev, 2019-08-07:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26810'>Zendesk ticket #26810</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

In order to better understand and help you solve the problem, need to know more information as follows:
1. Which version of Linux do you use? (recommended using Ubuntu16.04)
2. ROS version? (recommended using ROS-Kinetic)
3. GCC and Cmake versions? (GCC recommended 5.4.0, Cmake recommended 3.5.1)

In addition, it is recommended that you use OSDK-3.8.1. If this issue occurs, the solution is as follows:
You can switch off the function of ADVANCED_SENSING, and modify the code dji_sdk_node.cpp. The reference operation is as follows,
 [image]​
After the modification and saved, recompile it.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/YAELEqngSUAL4fpCAkKJVCtns' target='zd_attach'>inline-1750810702.png</a>

chaitanyantr, 2019-08-07:
ok as of now 3.8.1 is working for me.closing issue

closed 2019-08-07 by chaitanyantr

dji-dev, 2019-08-07:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26810'>Zendesk ticket #26810</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Congratulations on solving the problem and good luck with the project.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

MarkinYuriy, 2019-08-08:
hi, cannot finish catkin_make

[image]
[image]

Ubuntu 16.04.2 LTS \n \l
ros kinetic
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
on raspberry 3b+

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26810'>Zendesk ticket #26810</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The error prompted is the lack of related dependencies on the environment and advance sensing package. You need to install the related dependencies prompted. Or you can also change the camke file to -DADVANCED_SENSING to OFF.
 [image]​
Related dependencies tools:
 [image]​

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/D8ionYyvWGIsa4H5QWipTYv7x' target='zd_attach'>inline1321864295.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/BEG7hJDJGmWOzfyFE3AN7fgmS' target='zd_attach'>inline1534986568.png</a>

adeelahsaanawan, 2025-02-06:
I am using DJI OSDK 4.1.0 and DJI OSDK ROS 4.1.0 as well, but when after installing DJI OSDK, when I compile the DJI OSDK ROS in separate workspace, it gives me error: 

dji@manifold2:~/catkin_ws$ catkin_make
Base path: /home/dji/catkin_ws
Source space: /home/dji/catkin_ws/src
Build space: /home/dji/catkin_ws/build
Devel space: /home/dji/catkin_ws/devel
Install space: /home/dji/catkin_ws/install
####
#### Running command: "cmake /home/dji/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/dji/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/dji/catkin_ws/install -G Unix Makefiles" in "/home/dji/catkin_ws/build"
####
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dji/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/dji/catkin_nmea/devel;/opt/ros/kinetic
-- This workspace overlays: /home/dji/catkin_nmea/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dji/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - dji_osdk_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'dji_osdk_ros'
-- ==> add_subdirectory(Onboard-SDK-ROS-master)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Support Advanced Sensing
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/aarch64-linux-gnu/libusb-1.0.so
-- dji_osdk_ros: 38 messages, 75 services
-- USB INCLUDE: /usr/include/libusb-1.0

dji_osdk_ros...
-- Found OpenCV 3.3.1 installed in the system, will use it for depth perception sample
--  - Includes: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_viz;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dpm;opencv_face;opencv_fuzzy;opencv_hdf;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- Did not find ximgproc in OpenCV
-- Did not find viz3d in OpenCV
-- Found 9.0 CUDA installed in the system, will use it for depth perception sample
[... 891 lines cut ...]
[ 92%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node.cpp.o
[ 92%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node_mission_services.cpp.o
Onboard-SDK-ROS-master/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/build.make:62: recipe for target 'Onboard-SDK-ROS-master/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o' failed
make[2]: *** [Onboard-SDK-ROS-master/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/dji_vehicle_node_mission_services.cpp.o] Error 1
[ 92%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node_subscriber.cpp.o
[ 92%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node_publisher.cpp.o
CMakeFiles/Makefile2:6514: recipe for target 'Onboard-SDK-ROS-master/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/all' failed
make[1]: *** [Onboard-SDK-ROS-master/src/dji_osdk_ros/modules/CMakeFiles/dji_osdk_ros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 92%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node_mobile_comm.cpp.o
[ 93%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/modules/dji_sdk_node_payload_comm.cpp.o
[... cut ...]
[ 93%] Building CXX object Onboard-SDK-ROS-master/src/dji_osdk_ros_obsoleted/CMakeFiles/dji_sdk_node.dir/dji_linux_helpers.cpp.o
[ 94%] Linking CXX executable /home/dji/catkin_ws/devel/lib/dji_osdk_ros/dji_sdk_node
[ 94%] Built target dji_sdk_node
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

___________________________________________________________

Could anyone please share the link for both packages, which are compatible with each other.