dji-sdk/Onboard-SDK-ROS #278: Unable to 'catkin_make' for DJI_Onboard-SDK-ROS

AbhishekGS, 2019-08-08:
Hello all,

I am a newbie with ROS and DJI. I want to use DJI Onboard SDK ROS on my raspberry Pi. I installed ROS kinetic by following the instructions mentioned in the below link.

http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

I then followed the instructions in the below link to install DJI Onboard SDK and DJI Onboard SDK ROS.
http://wiki.ros.org/dji_sdk/Tutorials/Getting%20Started

The DJI onboard SDK is successfully installed in /home/<username>/DJI/DJI-Onboard-SDK directory.

The DJI Onboard SDK Ros is cloned in /home/<username>/catkin_ws/src directory.

But I am unable to execute 'catkin_make' from /home/<username>/catkin_ws/ directory because of the below error.

[image]

I then cloned 'Common_Msgs' from https://github.com/ros/common_msgs.git to the /home/<username>/catkin_ws/src directory by executing the below command.

git clone https://github.com/ros/common_msgs.git

I then tried to execute the 'catkin_make' command, but I get the below error.

[image]

Please help me on how to proceed further.

Thanks,
Abhishek

dji-dev, 2019-08-08:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26876'>Zendesk ticket #26876</a>:<br><br>Dear Abhishek,

Thank you for contacting DJI.

We recommend using ubuntu16.04, may I ask what system is installed on your raspberry pi?
In addition, the error is produced due to missing dependencies on the system you are trying to build this package. All dependencies are ROS packages and you can automatically install them using rodeo

    $ cd ~/catkin_ws
    $ rosdep install --from-paths src --ignore-src --rosdistro=indigo -y

After installing required dependencies, make sure to clean up build and devel directories and refresh for the current shell and try building it again (Assuming you are using bash):

    $ rm -rf build devel
    $ source ~/.bashrc
    $ catkin_make

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

AbhishekGS, 2019-08-09:
Hi Bruce, 

Thank you for responding.

I tried the suggestions you mentioned. Unfortunately, it didn't work. I still get the same error message.

I have a Raspberry Pi 3 running Raspbian Stretch OS and ROS Kinetic. The DJI-SDK and DJI-SDK-ROS are versions 3.8. According to this [link](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021), it suggests ROS kinetic to Debian Jessie. Therefore I installed ROS kinetic. 

Please suggest next steps.

Thanks,
Abhishek

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26876'>Zendesk ticket #26876</a>:<br><br>Dear Abhishek,

Thank you for contacting DJI.

Sorry that we have not tested raspberry PI 3, so we cannot reproduce and test your problem.
Your problems are primarily related to ROS environment dependencies, have noticed you use Debian Jessie, so it is recommended that you try the following:

    $ cd ~/catkin_ws
    $ rosdep install --from-paths src --ignore-src --rosdistro=indigo -y --os=debian:jessie

    $ rm -rf build devel
    $ source ~/.bashrc
    $ catkin_make

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle