dji-sdk/Onboard-SDK-ROS #279: cannot success finish catkin_make

MarkinYuriy, 2019-08-08:
hi, cannot finish catkin_make

image
image

Ubuntu 16.04.2 LTS \n \l
ros kinetic
gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
on raspberry 3b+

[image]
[image]

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26903'>Zendesk ticket #26903</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The error prompted is the lack of related dependencies on the environment and advance sensing package. You need to install the related dependencies prompted. Or you can also change the camke file to -DADVANCED_SENSING to OFF.
 [image]​
Related dependencies tools:
 [image]​

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

MarkinYuriy, 2019-08-09:
changed
until  failed
[image]

MarkinYuriy, 2019-08-09:
[image]

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26903'>Zendesk ticket #26903</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you delete all and recompile according to the OSDK development documentation (https://developer.dji.com/onboard-sdk/documentation/development-workflow/sample-setup.html#ros-onboard-computer). (git clone OSDK-ROS set -DADVANCED_SENSING  to OFF)
In addition, if an error occurs again, advised you can provide a complete log document because no specific error content was found in this log, it is not materially helpful for troubleshooting.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle