dji-sdk/Onboard-SDK-ROS #280: Unable to call rosservice 'drone_task_control'

AbhishekGS, 2019-08-09:
Hi Support Team,

I have installed ROS Melodic on Ubuntu 18.04 in a Virtual Machine(VM). I have installed DJI-SDK-ROS 3.7.0-5-g5325b65 and DJI-OSDK 3.7.0-3-gc3bb8b7. 

On Terminal 1, I do the following
`roslaunch dji_sdk sdk.launch`

On another terminal(Terminal 2), I do the following,
`rosservice call /dji_sdk/drone_task_control "task: 4" `

I get the below error message.

[image]

I have the DJI Assistant 2 running in Windows PC and the Simulator is on. I have the receiver (SBUS) connected to DJI N3 and Transmitter (Futaba T6K) is switched on. The Baud rate, App key, and App ID are set appropriately.

If I execute a subscribe command in terminal 2, as shown below, it executes successfully and I can see the IMU data on the screen.
`rostopic echo /dji_sdk/imu 
`
Please help me with executing the example rosservice command.

Thanks,
Abhishek

dji-dev, 2019-08-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26937'>Zendesk ticket #26937</a>:<br><br>Dear Abhishek,

Thank you for contacting DJI.

The recommended environment is ubuntu16.04 and ROS Kinetic, and suggest you use the recommended environment and DJI SDK ROS 3.8 for testing and validation.
 [image]​
Ubuntu16.04 (DJI manifold2-G) + ROS Kinetic + OSDK-ROS-3.8 + N3 can take off normally after our actual test, so recommended that you use the recommended environment and SDK version.
 [image]​

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

AbhishekGS, 2019-08-12:
Hi Bruce,

Thank you for responding.

I tried with the recommended versions as suggested. Below are the details
Software: Ubuntu 16.04.5 (LTS), ROS(Kinetic), DJI_ONBOARD_SDK_ROS(V3.8)
```
asheshad@ubuntu:~/dji_catkin_ws$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
```

```
asheshad@ubuntu:~/dji_catkin_ws/src/Onboard-SDK-ROS$ git describe --tags
3.8.1-2-gfe32f40
```

```
asheshad@ubuntu:~/dji_catkin_ws/src/Onboard-SDK-ROS$ rosversion -d
kinetic
```

Hardware: Only N3(That means I didn't assemble a drone, and not connected Compass as well)
Futaba T6K Transmitter and R3006SB SBUS receiver
Problem: The response result of service "drone_task_control' is always false.
Specifically, the output is

On Terminal 1:

```
asheshad@ubuntu:~/dji_catkin_ws/src/Onboard-SDK-ROS/dji_sdk$ roslaunch dji_sdk sdk.launch
... logging to /home/asheshad/.ros/log/f8b721d0-bcf1-11e9-a254-000c292328c9/roslaunch-ubuntu-37369.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38529/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1072802
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: 491b3222869b48821...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [37379]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8b721d0-bcf1-11e9-a254-000c292328c9
process[rosout-1]: started with pid [37392]
started core service [/rosout]
process[dji_sdk-2]: started with pid [37403]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0DBDF5E0020175

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x303082F

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
[ INFO] [1565608313.263188915]: drone activated
[ INFO] [1565608313.381370117]: Use data subscription to get telemetry data!
[ INFO] [1565608313.381464771]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 2 result: 0.
STATUS/1 @ startPackage, L352: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L350: Start package 1 result: 0.
STATUS/1 @ startPackage, L352: Package 1 info: freq=50, nTopics=14.

STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=5, nTopics=6.

STATUS/1 @ startPackage, L350: Start package 3 result: 0.
STATUS/1 @ startPackage, L352: Package 3 info: freq=400, nTopics=1.

STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

```
The last 2 rows of the output are:

```
STATUS/1 @ getErrorCodeMessage, L656: droneTaskCallback
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

```

In Terminal 2, I always get the result as False when I run the drone_task_control as shown below:

```
asheshad@ubuntu:~/dji_catkin_ws$ rosservice call /dji_sdk/drone_task_control "task: 4"
result: False
cmd_set: 1
cmd_id: 1
ack_data: 113

```

However, I can successfully subscribe to the IMU data on terminal 2, as shown below:

```
asheshad@ubuntu:~/dji_catkin_ws/src/Onboard-SDK-ROS$ rostopic echo /dji_sdk/imu
---
header: 
  seq: 476664
  stamp: 
    secs: 1565610705
    nsecs: 476841071
  frame_id: "body_FLU"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.707106781187
  w: 0.707106781187
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00891292933375
  y: -0.000242771755438
  z: 0.0019685907755
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.00888929514738
  y: -0.0795928315613
  z: 9.91668618
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

Could you please let me know how to proceed?

dji-dev, 2019-08-13:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26937'>Zendesk ticket #26937</a>:<br><br>Dear Abhishek,

Thank you for contacting DJI.

Sorry to have kept you waiting for so long. During this period, we have made many attempts but could not repeat your problem. The test result of the reply in the last email can work normally.
In addition, in order to better understand and help you solve the problem, we need to know more information as follows:
1. When executing roslaunch dji_sdk sdk.launch, what is the error probability of "getErrorCodeMessage, L656: droneTaskCallback"? 100% or other?
2. Directly run the flight control sample of OSDK-ROS, and what was the result?
3. What is ubuntu 16.04 running platform? PC-x86 or ARM onboard computer and model name?
4. Do you enable the "Enable SDK Failsafe Action box" in the DJI Assistant2 SDK Settings page?
5. And recommended to set the baud rate to 1M and verify again.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

rcabg, 2019-10-01:
Hello, I am having the same problem. My configuration is:

- DJI N3 autopilot with v1.7.7 firmware (newest) + Futaba (SBUS)
- OSDK v3.9
- Ubuntu 16.04 - x64 Upboard
- SDK API enabled + SDK Failsafe enabled + 1M baud rate configured

I only could make it work just after full autopilot reset. Once the drone is landed, you cannot take it off again, no matter if it is armed/disarmed or has the authority or not.

Any idea?
Thanks!

rcabg, 2019-10-01:
I've just discovered that SDK gives error when FUTABA remote controller is ON. If you keep the FUTABA off during operation it works, at least with the flight-control sample. Any suggestion?

a-for-apollo, 2019-11-01:
Hi, any news on the issue? I have the same error log but working on the M210.

dji-kyle, 2021-07-08:
Please ensure the RC stick back to center, and then controled by OSDK.

closed 2021-07-08 by dji-kyle