dji-sdk/Onboard-SDK-ROS #281: Why M600 pro can only fly to 30meter high in assistant2 simulator

pzy2253999075, 2019-08-10:
I try to control M600 pro in assistant2 simulator ，but it only can fly to 30 meter high in a very limited range.Could someone tell me why ? Thanks.

pzy2253999075, 2019-08-12:
@dji-dev could you tell me how to solve this  problem？Thanks

dji-dev, 2019-08-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26953'>Zendesk ticket #26953</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Do you get the same results with remote control as with OSDK?
It is recommended that you check whether the flight altitude limit is set in the DJI APP?
In addition, please confirm whether to open the remote control of M600 Pro?

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

pzy2253999075, 2019-08-13:
@dji-dev 你好，感谢你的回答。我试了一下，发现 只有 M600 PRO 的遥控器接上DJI GO APP后，才没有这个飞行限制，请问如何在不接App的情况下也消除限制？因为我只需要通过Onboard端控制。我还发现一个问题，当不开遥控器时，OnboardSDK好像控制不了无人机，例如我调用takeOff Service，模拟器中的M600桨只转了几下就不转了；以及在悬停过程中，我关掉遥控器，飞机也自己降落，请问有办法在不启动遥控器的情况下也可以用OnboardSDK控制无人机吗。谢谢

dji-dev, 2019-08-13:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26953'>Zendesk ticket #26953</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

如果您希望关闭遥控器也能使用OSDK进行控制，需要在DJI Assistant 2中勾选“Enable SDK Failsafe Action”选项。建议您先阅读DJI OSDK的开发文档，例如 https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/GQTH6q3VSvy2FJ7Y1eLUvchCM' target='zd_attach'>inline-1310328254.png</a>

closed 2021-07-08 by dji-kyle