dji-sdk/Onboard-SDK-ROS #284: Build Advanced Sensing error undefined reference to `cv::cuda::createStereoBM(int, int)'

diogon01, 2019-08-14:
I try build the example in Jetson TX2 and not work

```
!#shell
CMakeFiles/demo_stereo_depth_perception.dir/src/stereo_utility/stereo_frame.cpp.o: In function `M210_STEREO::StereoFrame::initStereoParam()':
/home/nvidia/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_demo/src/stereo_utility/stereo_frame.cpp:75: undefined reference to `cv::cuda::createStereoBM(int, int)'
CMakeFiles/demo_stereo_depth_perception.dir/src/stereo_utility/stereo_frame.cpp.o: In function `M210_STEREO::StereoFrame::rectifyImgs()':
/home/nvidia/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_demo/src/stereo_utility/stereo_frame.cpp:162: undefined reference to `cv::cuda::remap(cv::_InputArray const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&, int, int, cv::Scalar_<double>, cv::cuda::Stream&)'
/home/nvidia/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_demo/src/stereo_utility/stereo_frame.cpp:169: undefined reference to `cv::cuda::remap(cv::_InputArray const&, cv::_OutputArray const&, cv::_InputArray const&, cv::_InputArray const&, int, int, cv::Scalar_<double>, cv::cuda::Stream&)'
collect2: error: ld returned 1 exit status
```

My CmakeText call OpenCV for ros:

```
!#shell
/usr/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /home/nvidia/catkin_ws/src
-- Using CATKIN_DEVEL_PREFIX: /home/nvidia/catkin_ws/src/cmake-build-debug/devel
-- Using CMAKE_PREFIX_PATH: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/nvidia/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nvidia/catkin_ws/src/cmake-build-debug/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - nmea_msgs
-- ~~  - dji_sdk
-- ~~  - dji_sdk_demo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'nmea_msgs'
-- ==> add_subdirectory(nmea_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- nmea_msgs: 6 messages, 0 services
-- +++ processing catkin package: 'dji_sdk'
-- ==> add_subdirectory(Onboard-SDK-ROS/dji_sdk)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using Finddjiosdk-core.cmake find djiosdk-core
-- Found djiosdk-core:
--  - Includes: /usr/local/include
--  - Libraries: /usr/local/lib/libdjiosdk-core.a
-- dji_sdk: 13 messages, 28 services
-- +++ processing catkin package: 'dji_sdk_demo'
-- ==> add_subdirectory(Onboard-SDK-ROS/dji_sdk_demo)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found OpenCV 3.3.1 installed in the system, will use it for depth perception sample
--  - Includes: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
-- Found ximgproc module in OpenCV, will use it to filter disparity map in depth perception sample
-- Found viz3d in OpenCV, will use it to visualize point cloud
-- Found 9.0 CUDA installed in the system, will use it for depth perception sample
--  - Includes: /usr/local/cuda-9.0/include
-- Found Advanced Sensing in the system
-- Found OpenCV 3.3.1, Viz3d, and advanced sensing module, depth percpetion demo will be compiled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/catkin_ws/src/cmake-build-debug

[Finished]

```

dji-dev, 2019-08-14:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27052'>Zendesk ticket #27052</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Recommended that you provide more complete compilation log information, and what version of OSDK are you using? OSDK-3.8.1？
Please confirm the version of your environment dependency tool as follows:
 [image]​

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/xihKvrh01suSxvS2komQcD2fS' target='zd_attach'>inline-1330518317.png</a>

diogon01, 2019-08-17:
Thank`s for re-ward, Yes I`m using branch OSDK-3.8.1
My enviroment:
```
!#shel
~$ gcc --version

gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
~$ cmake --version
cmake version 3.5.1
~$ pkg-config --modversion opencv
3.3.1

```
All information about my OpenCV and respective modules

```
#!shell
General configuration for OpenCV 3.3.1 =====================================
  Version control:               3.3.1

  Extra modules:
    Location (extra):            /home/nvidia/third-party-library-folder/opencv_contrib/modules
    Version control (extra):     3.3.1

  Platform:
    Timestamp:                   2019-08-09T19:41:39Z
    Host:                        Linux 4.4.38-tegra aarch64
    CMake:                       3.5.1
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON
      disabled:                  VFPV3

  C/C++:
    Built as dynamic libs?:      YES
    C++ Compiler:                /usr/bin/c++  (ver 5.4.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wno-narrowing -Wno-delete-non-virtual-dtor -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wno-narrowing -Wno-delete-non-virtual-dtor -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-narrowing -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-narrowing -Wno-comment -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):
    Linker flags (Debug):
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          dl m pthread rt /usr/lib/aarch64-linux-gnu/libGLU.so /usr/lib/aarch64-linux-gnu/libGL.so /usr/lib/aarch64-linux-gnu/libtbb.so cudart nppc nppial nppicc nppicom nppidei nppif nppig nppim nppist nppisu nppitc npps cublas cufft -L/usr/local/cuda-9.0/lib64
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 cudev core cudaarithm flann hdf imgproc ml objdetect phase_unwrapping plot reg surface_matching video viz xphoto bgsegm cudabgsegm cudafilters cudaimgproc cudawarping dnn face freetype fuzzy img_hash imgcodecs photo shape videoio xobjdetect cudacodec highgui ts bioinspired dpm features2d line_descriptor saliency text calib3d ccalib cudafeatures2d cudalegacy cudaobjdetect cudaoptflow cudastereo cvv datasets rgbd stereo structured_light superres tracking videostab xfeatures2d ximgproc aruco optflow stitching python2 python3
    Disabled:                    js world contrib_world
    Disabled by dependency:      -
    Unavailable:                 java cnn_3dobj dnn_modern matlab sfm

  GUI: 
    QT 5.x:                      YES (ver 5.5.1)
    QT OpenGL support:           YES (Qt5::OpenGL 5.5.1)
    OpenGL support:              YES (/usr/lib/aarch64-linux-gnu/libGLU.so /usr/lib/aarch64-linux-gnu/libGL.so)
    VTK support:                 YES (ver 6.2.0)

  Media I/O: 
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.8)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver )
    WEBP:                        /usr/lib/aarch64-linux-gnu/libwebp.so (ver encoder: 0x0202)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.2.54)
    TIFF:                        /usr/lib/aarch64-linux-gnu/libtiff.so (ver 42 - 4.0.6)
[... 74 lines cut ...]

  Python (for build):            /usr/bin/python2.7

  Java:
    ant:                         NO
    JNI:                         NO
    Java wrappers:               NO
    Java tests:                  NO

  Matlab:                        Matlab not found or implicitly disabled

  Documentation:
    Doxygen:                     NO

  Tests and samples:
    Tests:                       YES
    Performance tests:           YES
    C/C++ Examples:              YES

  Install path:                  /usr/local

  cvconfig.h is in:              /home/nvidia/third-party-library-folder/opencv/build
-----------------------------------------------------------------

```

dji-dev, 2019-08-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27052'>Zendesk ticket #27052</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to the error log, the issue may be Cuda version or environment configuration.
So, it is recommended that you can uninstall the Cuda, then try to build the OSDK again.
Or install the Cuda version to 8.0, and deal with the dependencies.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

potBagMeat, 2019-10-08:
any further information?
I confronted the same question as you do, but I had cuda 9.0 installed in my Manifold 2-G.
did the question solved?
I will really appreciate it  if you answer me.

diogon01, 2019-10-08:
> any further information?
> I confronted the same question as you do, but I had cuda 9.0 installed in my Manifold 2-G.
> did the question solved?
> I will really appreciate it if you answer me.

Yes man i solved! The probloblem because have conflit in OpencV ROS i have two solutions for you:

Install ROS base only (don't forgive install dependencies manually) :
`sudo apt-get install ros-kinetic-ros-base`

OR edit the cmake file in line 280

`gedit ~/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_demo/CMakeLists.txt`

```C++
if (OPEN_CV_3_3_0_INSTALLED AND FOUND_OPENCV_VIZ AND darknet_ros_FOUND AND FOUND_ADVANCED_SENSING)
    message(STATUS "Found OpenCV ${OpenCV_VERSION}, Viz3d, darknet_ros, and advanced sensing module, depth percpetion demo will be compiled")

    set(OpenCV_INCLUDE_DIRS
            /usr/local/include
            /usr/local/include/opencv2
            )

    set(OpenCV_LIB_DIR
            /usr/local/lib
            )

    add_executable(demo_stereo_object_depth_perception
            src/demo_stereo_object_depth_perception.cpp
            ${STEREO_UTILITY_SRC_FILE} )
    target_link_libraries(demo_stereo_object_depth_perception
            ${catkin_LIBRARIES}
            ${DJIOSDK_LIBRARIES}
            ${OpenCV_LIBS}
            )
    add_dependencies(demo_stereo_object_depth_perception dji_sdk_generate_messages_cpp)
endif() 

```
And build again.

potBagMeat, 2019-10-08:
> > any further information?
> > I confronted the same question as you do, but I had cuda 9.0 installed in my Manifold 2-G.
> > did the question solved?
> > I will really appreciate it if you answer me.
> 
> Yes man i solved! The probloblem is result of a conflit in OpencV ROS i have two solutions for you:
> 
> Install ROS base only (don't forgive install dependencies manually) :
> `sudo apt-get install ros-kinetic-ros-base`
> 
> OR edit the cmake file in line 280
> 
> `gedit ~/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_demo/CMakeLists.txt`
> 
> ```c++
> if (OPEN_CV_3_3_0_INSTALLED AND FOUND_OPENCV_VIZ AND darknet_ros_FOUND AND FOUND_ADVANCED_SENSING)
>     message(STATUS "Found OpenCV ${OpenCV_VERSION}, Viz3d, darknet_ros, and advanced sensing module, depth percpetion demo will be compiled")
> 
>     set(OpenCV_INCLUDE_DIRS
>             /usr/local/include
>             /usr/local/include/opencv2
>             )
> 
>     set(OpenCV_LIB_DIR
>             /usr/local/lib
>             )
> 
>     add_executable(demo_stereo_object_depth_perception
>             src/demo_stereo_object_depth_perception.cpp
>             ${STEREO_UTILITY_SRC_FILE} )
>     target_link_libraries(demo_stereo_object_depth_perception
>             ${catkin_LIBRARIES}
>             ${DJIOSDK_LIBRARIES}
>             ${OpenCV_LIBS}
>             )
>     add_dependencies(demo_stereo_object_depth_perception dji_sdk_generate_messages_cpp)
> endif() 
> ```
> 
> And build again.

Hey,thank you, man.
Fortunately, I solved this problem right after I posted the question.
The way I did is that add a simple line in line 2 of the CMakeLists.txt of the dji_sdk_demo.
`cmake_minimum_required(VERSION 2.8.3)
set(OpenCV_DIR "/media/dji/f044edfd-716b-4b17-9a54-7a7cbc0e8505/third-party-library-folder/opencv/build")
project(dji_sdk_demo)`
Thanks for your solution anyway.

closed 2021-07-08 by dji-kyle