dji-sdk/Onboard-SDK-ROS #285: Please publish camera parameters

SwiftGust, 2019-08-19:
I'm using M210 RTK and there are 5 available cameras
down front/back, front left/right & fpv camera 

I could get those image streams in 240 x 320 resolution as in the documentation, 
however also I need camera parameters for use it to detect markers and so on.
Current OSDK ROS does not have camera parameter topic available, which makes much less useful. 

It would be wonderful if parameters of those cameras are readily available in OSDK ROS without further, separate calibration processes.

plus, I want to know how depth image (/dji_sdk/stereo_depth_image) should be scaled to be real metric? 

Thank you.

dji-dev, 2019-08-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27227'>Zendesk ticket #27227</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

If you have requirements for relevant functionality, It is recommended that you click on the links on the right side of the fill in the relevant demand information, https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements ( camera parameter topic? It can be more specific. What are the parameters? )
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

plus, I want to know how depth image (/dji_sdk/stereo_depth_image) should be scaled to be real metric?
==>It is recommended that you can refer to the "Object Depth Perception in Stereo Image" sample of  DJI Onboard SDK.
https://developer.dji.com/onboard-sdk/documentation/sample-doc/advanced-sensing-object-detection.html#sample-2-object-depth-perception-in-stereo-image

​Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

SwiftGust, 2019-08-20:
I submitted form now but leave similiar comment here to clarify.

What I want is nothing different from ROS standard, camera parameters as defined in [sensor_msgs/CameraInfo](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/CameraInfo.html) 
to reconstruct 3D pose or depth from images from front, downward vision sensors. 

Or it would be much better if vision based precision landing is available to M210 or other DJI drones.

And stereo depth samples comes with additional, separate calibration, tuning and the onboard processing required to front stereo image in VGA resolution which takes significant resources.
What I need is how to get depth in metric from 8bit 240p front disparity image.

dji-dev, 2019-08-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27227'>Zendesk ticket #27227</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thanks for your feedback and suggestion, and we will evaluate the requirement information you submit.
If you submit information, can you tell me the email address?

​Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

alessandroscoppio, 2020-03-04:
Any update on this? Are camera parameters available in a topic?