dji-sdk/Onboard-SDK-ROS #286: how to use  callback in onboardSdk

pzy2253999075, 2019-08-20:
@dji-dev  I want to know how to use  setWaypointEventCallback（），could someone give me  a demo  with  some code  ，thanks。Could you help me  。

dji-dev, 2019-08-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27250'>Zendesk ticket #27250</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

You can call setWaypointEventCallback to register the callback function. The code sample is as below.
 [image]​
In ROS you can add "setWaypointEventCallback" in the end of the function DJISDKNode::missionWpUploadCallback.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/598VcC3lGx4hojN0t3AwjW9oc' target='zd_attach'>inline-2085990225.png</a>

pzy2253999075, 2019-08-20:
@dji-dev got it ,thanks

dji-dev, 2019-08-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27250'>Zendesk ticket #27250</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It's our pleasure to answer your questions, good luck with the project.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle