dji-sdk/Onboard-SDK-ROS #288: 关于摄像头信息以Ros的方式获取

pzy2253999075, 2019-08-20:
@dji-dev  你好，M600 pro上装的禅思系列云台相机的图像可以通过Ros话题的形式获取吗，OnboardSdkROS有这样的接口吗？谢谢。

dji-dev, 2019-08-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

很抱歉，DJI Onboard SDK不支持获取M600 Pro的摄像头图像，目前仅支持M210系列。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-21:
@dji-dev 感谢回答，今天我测试了一下OnboardSdkRos的waypointMission，发现有个问题：我在assistant2中模拟M600的巡航任务，遥控器是关闭状态，点上传任务没有问题，但是点开始执行的时候显示“FAR AWAY FROM HOME”，但是航点离起飞点都很近。然后打开遥控器，也还是同样的报错，直到插上手机打开DJI GO 才成功起飞执行任务。飞机返航后我又关闭遥控器，看看能不能再起飞执行任务，发现第二次没有遥控器和app也都能执行。我猜测是：只有连上DJI GO才把assitant2中模拟器的起点给无人机，还是说因为其他原因呢？如果在空旷Gps好的真实环境下，我第一次直接用onboardSDK（不打开遥控器和app）让无人机执行waypointMission，还是会报这样的错误吗？谢谢

因为没有抢权限，所以会报以下错误，onboardsdk抢了权限就能成功上传了
[image]

没打开app前点开始执行会报以下错误，打开app后就不报错并成功执行，并且后续关闭app和遥控器也能成功执行上传任务和开始任务
[image]

pzy2253999075, 2019-08-22:
@dji-dev 请问如何解决呢

dji-dev, 2019-08-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

为了更好的了解和帮助您解决问题，建议您根据下方提示步骤进行问题排查，并点击右侧的超链接填写和提交排查反馈信息， https://formcrafts.com/a/dji-developer-feedback-cn
1. 请问您是直接运行的OSDK mission sample？
2. 如果不是，建议尝试运行DJI OSDK-3.8.1 的 mission sample ，运行waypoint，验证是否还会出现该问题
3. 如果问题继续发生，建议提供完整的程序log，以及相应截图或者操作视频
4. M600 Pro的固件版本
5. 问题出现的概率， 100%还是其他？

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-23:
@dji-dev 运行demo_mission是没问题的，没有遥控器也可以上传任务起飞，我再研究下，谢谢

pzy2253999075, 2019-08-23:
@dji-dev 你好 ，我又测试了一遍demo_mission，因为demo_mission原来的航点之间距离很短，只有不到10m，直接上传并执行任务并没有报错，也不需要打开遥控器。而在demo_mission中换成我自己的点后，每个航点之间的距离在50到60m，上传没问题，但开始执行就报”MISSION_TOO_FAR_FROM_HOME“,打印的error_code如下：
[image]
只有打开遥控器并脸上app才可以开始执行，只要第一次执行成功后，后面关闭遥控器，也可以继续上传并执行，也没报错。请问怎么解决呢，即第一次我不开遥控器，也能上传并执行，谢谢

dji-dev, 2019-08-23:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

为了更好的了解和帮助您解决问题，建议您点击右侧的超链接填写和提交更完整的问题信息， https://formcrafts.com/a/dji-developer-feedback-cn
其中问题描述栏尽量详细，包括如下内容：
1. 使用的OSDK版本
2. M600 Pro的固件版本
3. 程序运行或者测试的完整log（可以添加到附件）
4. 您的测试程序（可以添加到附件）
5. 如果需要尝试复现您的问题，建议简要提供完整操作步骤

感谢您的理解与支持,祝您生活愉快!

Best Regards，

pzy2253999075, 2019-08-25:
@dji-dev 
1、OnboardSdk版本是3.8.1版本
2、M600 pro 的固件版本是V1.00.01.67
3、Log就是上上个框里面的图片
4、测试程序就是在把OnboardSdkRos里面的demo_mission.cpp里面的默认矩形的航点，换成我们自定义的一些航点，这些航点距离我们在assistant2 simulator中设置的无人机仿真坐标都是差不多50到100m之间。（其他程序都没改）
或者我可以给出这些点：
assistant 2  simulator  中的点是这个 22.5298494456，113.931877563
[image]

waypointMission的航点如下4个点
22.5291359074，113.9327251411
22.5296066448，113.9336961007
22.5304440579，113.9346456027
22.5314102975，113.9341306186

dji-dev, 2019-08-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

已经收到您提供的信息，我们需要对您遇到的问题做进一步的分析和验证，一有结果我们将第一时间通知您。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-dev, 2019-08-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

经过实际验证和测试无法复现您的问题，建议您检查首个航点位置和模拟器位置是否有偏差，如果有，可以将第一个航点位置修改为模拟器的位置，再进行验证。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-31:
@dji-dev 好的，我再试试

dji-dev, 2019-09-02:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

很高兴能为您解答问题，祝您项目顺利!

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-09-05:
@dji-dev 你好，我已经测试过了，第一次完全不打开遥控器的情况下，而且第一个航点和无人机在模拟器上的点一模一样，还是报了MISSION_TOO_FAR_FROM_HOME  ack.data = 199。不知怎么解决。。。

dji-dev, 2019-09-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

很抱歉该问题给您造成的困扰，我们需要对该问题重新做进一步的分析和验证，一有结果我们将第一时间通知您。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-dev, 2019-09-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27286'>Zendesk ticket #27286</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

经过进一步的验证和确认，需要开启遥控器连接DJI GO进行初始化信息认证，否则只能在半径为50m高为30m的圆柱形范围内活动，需要保证每个航点都在这个范围内。
建议您可以先开启遥控器连接DJI GO（进行初始化信息认证），再选择关闭，这样可以取消这样的范围限制。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-09-09:
@dji-dev 好 谢谢

closed 2021-07-08 by dji-kyle