dji-sdk/Onboard-SDK-ROS #290: How to get the drone heading

chaitanyantr, 2019-08-21:
How to get the drone heading in ROS,i want the drone heading info for the rotation matrix from FLU to ENU frame.

I'm sending the velocity commands in body frame of drone, but the local position i'm getting is with respect to ENU frame.

chaitanyantr, 2019-08-26:
Thanks for response,but to get clarity what is the values i'm getting  from attitude yaw..!!..is it change in drone yaw with respect to ENU frame.?

closed 2019-08-28 by chaitanyantr

waltermayor, 2020-02-27:
HI
I have a matrice 600 and I am trying to use the ros package robot_localization  to obtain the pose of my drone. the robot_localization package need IMU data in ENU frame, but the topic /dji_sdk/imu from the drone is in FLU frame.

How can I set my IMU data from FLU to ENU?