dji-sdk/Onboard-SDK-ROS #291: 关于OnboardSdk  低电量返航以及Mission 低电量returnHome  API

pzy2253999075, 2019-08-23:
@dji-dev 你好，请问OnboardSdk可以像MobileSdk一样设置无人机低电量返航，以及可以设置mission任务智能返航："判断当前电量是否能够刚足以返航，如果是就返航"  的API吗，我在api文档里面找不到相关啊函数。

dji-dev, 2019-08-23:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27406'>Zendesk ticket #27406</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

1. Onboard SDK没有设置无人机低电量返航的API，不过建议您可以使用DJI APP进行设置，或者使用OSDK监测电池电量状态。
2. mission 任务是没有智能返航功能的，智能返航功能是设置在飞控上，比如您提到Mobile Sdk的Smart go home API。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-25:
@dji-dev 你好，我们测试过通过DJI APP 设置低电量返航，如果遥控器失联或者被关掉，无人机的电量即使低于低电量返航的阈值，也还是继续执行任务（但打开遥控器后能返航），这种情况下是否只能通过我们额外实现监测代码来让无人机返航？还有第二点，你们说mission任务没有只能返航的api，那OnboardSdk是否支持智能返航模式？或者说M600的A3飞控是否已经默认设置智能返航功能？如果没有，是否只能通过用户自己实现代码判断来实现自能返航？我担心如果用户自己实现逻辑判断，假如计算不精准会导致无人机安全性问题。谢谢回答

pzy2253999075, 2019-08-26:
@dji-dev 可以回答我的问题吗，谢谢

dji-dev, 2019-08-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27406'>Zendesk ticket #27406</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

1. 我们测试过通过DJI APP 设置低电量返航，如果遥控器失联或者被关掉，无人机的电量即使低于低电量返航的阈值，也还是继续执行任务（但打开遥控器后能返航），这种情况下是否只能通过我们额外实现监测代码来让无人机返航？
==> 您可以使用智能返航功能（DJI APP中设置开启），或者使用JI Onboard SDK监测电池电量，然后调用OSDK的返航API。

2. 还有第二点，你们说mission任务没有只能返航的api，那OnboardSdk是否支持智能返航模式？或者说M600的A3飞控是否已经默认设置智能返航功能？
==> Onboard SDK没有智能返航模式，但M600具备智能返航功能，建议您使用前在APP中确认是否开启，关于更多功能细节，建议您联系 support@dji.com 会得到更专业的支持

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-08-27:
@dji-dev 请问DJI APP的低电量智能返航是指：无人机在当前位置的电量如果刚好够返航就返航是吗，如果是，那么在APP上设置了之后，然后关掉遥控器和APP，此设置还成立吧？还可以在OnboardSdk的mission过程中生效吗。

pzy2253999075, 2019-08-31:
@dji-dev 好的 感谢回答

dji-dev, 2019-09-02:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27406'>Zendesk ticket #27406</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

很高兴能为您解答问题，祝您项目顺利!

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-07-08 by dji-kyle