dji-sdk/Onboard-SDK-ROS #293: Update to 3.9 for ROS Package?

InterestingWalrus, 2019-10-08:
Hi, do you have an ETA for when the ROS version of the Onvoard-SDK will be updated to V3.9?

dji-dev, 2019-10-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28644'>Zendesk ticket #28644</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that due to the uncertainty of software development time, I cannot guarantee you the specific release time of OSDK-ROS 3.9.
But if you have requirements about the functionality of OSDK-ROS 3.9, It is recommended that you click on the links on the right side of the fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

InterestingWalrus, 2019-10-09:
Thanks Bruce, the functionality I'm interested have been addressed in OSDK-3.9. They are the battery bug and the RTH function. I'll wait for v3.9 of the ROS driver I guess.

closed 2019-10-09 by InterestingWalrus

dji-dev, 2019-10-10:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28644'>Zendesk ticket #28644</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team