dji-sdk/Onboard-SDK-ROS #296: Odd data on topic /dji_sdk/imu (linear_acceleration)

erenaud3, 2019-10-11:
Hi !

I just generated a bag file thanks to a very simple simulation. I am only performing a take off, some "random" displacement (playing around with the joystick) and then a return-to-home.

If I take a look at /dji_sdk/imu's documentation, I can read: 

> IMU data including raw gyro reading in FLU body frame, raw accelerometer reading in FLU body frame, and attitude estimation, published at 100 Hz for M100, and 400 Hz for other platforms. Note that raw accelerometer reading will give a Z direction 9.8 m/s2 when the drone is put on a level ground statically. 

So obviously, raw acceleration on X and Y axis should be 0 and Z should be 9.8 when drone is landed, right ? Well, I don't. Here's the data I am getting (topic /dji_sdk/imu and data is linear_acceleration) :

[image]

Here's the Z axis' data on /dji_sdk/height_above_takeoff, so you have an idea of when the drone is landed, taking-off or landing.

[image]

So why I am getting this data ? For example, linear acceleration on Y axis can't be constant at 5m/s2 during the whole flight, that's just silly. Is that because data produced by simulation aren't reliable ?

Thank you.

dji-dev, 2019-10-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28764'>Zendesk ticket #28764</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

To better understand and help you solve the problem, we need to know more about the following information:
1. Are you using DJI Assistant2 for simulation?
2. What is the specific model of DJI aircraft you are using? And the firmware version?
3. The version of DJI OSDK you are using
4. The bag file generated during test
5. Flight record of simulator test (exported from DJI Assistant2)

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-14:
Thank you for your reply.

1. Yes, I am. "DJI Assistant2 For Matrice" to be more specific.
2. I am using a M600. Firmware version is 1.0.1.67.
3. DJI OSDK version is 3.8.1.
4. [Bag file](https://drive.google.com/open?id=1QeYzPoteMNNrPF2CdP8Jyv1t0jTGItXg)
5. [Flight data](https://drive.google.com/file/d/10n8UHiJydysLgQCmU52FeEJmJP6NIkmI/view?usp=sharing)

One thing: for the first bag file, I started to record AFTER launching ROS node (command roslaunch dji_sdk sdk.launch). For the bag file I am posting right now, I started to record BEFORE.

dji-dev, 2019-10-15:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28764'>Zendesk ticket #28764</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the information you provided. We will make further analysis and verification of the problems you raised and will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2020-03-02:
Dear @erenaud3 ,
could you resolve your problem? 
By checking your bag files, I realized, that your IMU publishing rate behave same like mine in [issue #325](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/325). Your IMU is always publishing in different speeds (at least ROS bag is recording it like this). Did you had a look on this too?

Best regards,

Max

dji-dev, 2020-07-10:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28764'>Zendesk ticket #28764</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to keep you waiting, OSDK has been updated to 4.0, and the test on OSDK4.0 and OSDK-ROS 4.0.1(master branch) show normal results, the z-axis acceleration at rest is about 9.8. It is recommended to use the latest SDK version for development and validation.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

tkazik, 2020-09-22:
Dear DJI SDK Support

Could you please double check your answer?
To my understanding, the **M600/Pro** is only supported by OSDK 3.x and OSDK-ROS 3.x.
Unfortunately, version 4.x is only available for newer models (M210v2 or M300).
So your suggestion does **not** solve this issue.