dji-sdk/Onboard-SDK-ROS #297: What's the unit of the angle when using attitude control?

sunhaoshui, 2019-10-12:
I am using the angle control api(horizontal angle) to control the a3 plane. I am curious about the unit of the angle. I thought it was in deg(the document said the limit of the input is 35deg).But it seems it is in rad when I use dji assitance to simulate it.

dji-dev, 2019-10-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28775'>Zendesk ticket #28775</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

About the horizontal Angle control, the flightCtrl API ends up a unit of use is the Angle, but if you are using DJI OSDK "DJISDKNode ROS: : flightControl" interface, the unit is the radian, can refer to the code content (figure below)
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/gnf98lwwAMtMyUDsKLfX5Mdku' target='zd_attach'>inline885831465.png</a>

sunhaoshui, 2019-10-12:
ok got it. Thanks a lot!

closed 2019-10-12 by sunhaoshui

dji-dev, 2019-10-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28775'>Zendesk ticket #28775</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

​It's our pleasure to answer your questions, good luck with your project.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team