dji-sdk/Onboard-SDK-ROS #298: Hardware synchronisation implementation

erenaud3, 2019-10-14:
Hello,

I am currently working on a M600 and I would like to implement hardware synchronization. So I read this [page on the documentation (Hardware Sync )](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-hardware-sync.html) and spent a lot of times looking at the code but I am still not quite sure I fully understand. To be more specific, I am still uncertain about what is already implemented in this ROS package and what I would have to do myself.

So if I understand correctly, the flight controller will first send a digital pulse from FC (flight controller) to OC (Onboard Computer) through hardware line. At the same time, it  will also sent a HardSyncData structure over UART that will be received by ROS. Then ROS will publish the received data on topic /dji/trigger_time.

And that's all for what is already implemented, am I correct ?

Now, let's take a look at what I would have to do myself. First, I would need to right some piece of codes (let's say a ROS node) that would catch the digital pulse sent on hardware line in order to get arrival time. Then, this node would have to wait for /dji/trigger_time topic to get time at which the pulse was sent by Flight Controller. Finally, the node would only have to solve the equation in the link I posted above to perform hardware synchronization.

`(t_OC) = (t0_OC) + (t_A3) - (t0_A3) 

t_OC : current clock time in OC
t0_OC : OC's clock time at which pulse was sent
t_A3 : current clock time in A3
t0_A3 : A3's clock time at which pulse was received
`

Am I still correct ? 

One last question, does anybody would have any idea about how I could check that data fusing is more efficient with hardware synchronization than without it ?

Thank you !

dji-dev, 2019-10-15:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28841'>Zendesk ticket #28841</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

And that's all for what is already implemented, am I correct ?
==> Yes, your understanding is correct

Am I still correct?
==> Your understanding seems to be ok, and is there anything else you don't understand about using?

One last question, does anybody would have any idea about how I could check that data fusing is more efficient with hardware synchronization than without it ?
==> DJI A3/N3 hardware synchronization time-aligned Data only has an IMU, and if you need to use IMU Data for related processing and calculation, using hardware synchronization is more efficient.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-15:
Thank you for your answer.

> Am I still correct?
> ==> Your understanding seems to be ok, and is there anything else you don't understand about using?

Actually, there is. If everything I stated is correct, I think I don't quite understand what is the purpose of functions DJISDKNode::alignRosTimeWithFlightController(...) and DJISDKNode::publish400HzData(...) as it is out of the box (I mean without some kind of hardware synchronization mechanism that I implemented myself).

I enabled time alignment in sdk.launch to test it like this:
```
<param name="align_time" type="bool" value="true"/>
```
So my additional questions:

---

1. Does time alignment have anything to do with hardware synchronization ?
From what I understood, **time alignment doesn't require hardware synchronization to work**, but data fusing will be possible only if hardware synchronization is enabled AND if ROS messages are timestamped with Flight Controller clock (i.e, time alignment is enabled as well). Am I correct ?

---

2. What is  DJISDKNode::alignRosTimeWithFlightController(...) actually doing ?
I feel like it only prevent messages to be timestamped with FC's clock until we are pretty sure that ROS' and FC's clocks are "synchronized". 

---

 In DJISDKNode::publish400HzData(...) :
```cpp
void
DJISDKNode::publish400HzData(Vehicle *vehicle, RecvContainer recvFrame,
                                  DJI::OSDK::UserData userData)
{

  // (...)

  if(p->align_time_with_FC)
  {
    p->alignRosTimeWithFlightController(now_time, packageTimeStamp.time_ms);
    if(p->curr_align_state == ALIGNED)
    {
      msg_time = p->base_time + _TICK2ROSTIME(packageTimeStamp.time_ms);
    }
    else
    {
      return;
    }
  }
```

3. What is exactly p->base_time ? 

4. How are packageTimeStamp done ? Is that number of ticks since 1970 (for instance) or number of ticks since beginning of current flight ?

dji-dev, 2019-10-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28841'>Zendesk ticket #28841</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

1. Does time alignment have anything to do with hardware synchronization ? From what I understood, time alignment doesn't require hardware synchronization to work, but data fusing will be possible only if hardware synchronization is enabled AND if ROS messages are timestamped with Flight Controller clock (i.e, time alignment is enabled as well). Am I correct ?
==> OSDK Time synchronization function is applicable to M210 V2, and the OSDK hardware synchronization function is applicable to A3/N3. The ultimate purpose is to achieve the synchronization of the two systems in time and data. Time synchronization also requires a synchronization signal on the hardware.

2. What is DJISDKNode::alignRosTimeWithFlightController(...) actually doing ? I feel like it only prevent messages to be timestamped with FC's clock until we are pretty sure that ROS' and FC's clocks are "synchronized".
3. What is exactly p->base_time ?
==> The purpose of this function is to calculate the time difference between the flight control and ROS systems, and to ensure that the time difference is almost within "TIME_DIFF_CHECK", then you get relatively accurate base_time.
So, the p->base_time is an alignment reference time of the ROS and flight control system, it is also the time when flight control started to run (expressed by ROS system time).

4. How are packageTimeStamp done ? Is that number of ticks since 1970 (for instance) or number of ticks since beginning of current flight ?
==> This timestamp refers to the time after the flight controller is powered up.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-22:
Thank you very much for those useful information !

>     1. ==> OSDK Time synchronization function is applicable to M210 V2, and the OSDK hardware synchronization function is applicable to A3/N3. The ultimate purpose is to achieve the synchronization of the two systems in time and data. Time synchronization also requires a synchronization signal on the hardware.

But I still have one question to be crystal clear about this part of your answer. Are you saying that time alignment is the implementation of software synchronization ?

dji-dev, 2019-10-23:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28841'>Zendesk ticket #28841</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

But I still have one question to be crystal clear about this part of your answer. Are you saying that time alignment is the implementation of software synchronization?
==> This refers to the time synchronization of M210 V2 (PPS synchronization signal is required), which can be referred to https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-time-sync.html

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle