dji-sdk/Onboard-SDK-ROS #299: ESC Data Topic

ben-isaac, 2019-10-14:
Hi there,

I would like to add a new topic for reading the ESC data (motor speed, current, etc.) for DJI Matrice 210. I took the following steps for adding the topic in the ROS package. However, when I echo the topic it doesn't read the ESC data from the drone. I was wondering if you can please advise what the problem might be? Thank you.
1. Added ros::Publisher esc_publisher;	 in dji_sdk_node.h
2. Created a new message 'ESCStatus' as below
[image]
 
3. Added the 'ESCStatus.msg' in the CMakeList.txt
4. Added `esc_publisher = nh.advertise<dji_sdk::ESCStatus>("dji_sdk/esc",10); ` under the `DJISDKNode::initPublisher(ros::NodeHandle& nh)
{`

4. Added the following in the dji_sdk_node_publisher.cpp
` int i = 3;
  Telemetry::TypeMap<Telemetry::TOPIC_ESC_DATA>::type esc_data =
    vehicle->subscribe->getValue<Telemetry::TOPIC_ESC_DATA>();
  dji_sdk::ESCStatus esc_status;
  esc_status.header.stamp = msg_time;
  esc_status.current = esc_data.esc[i].current;
  esc_status.speed = esc_data.esc[i].speed;
  esc_status.voltage = esc_data.esc[i].voltage;
  esc_status.temperature = esc_data.esc[i].temperature;
  esc_status.stall = esc_data.esc[i].stall;
  esc_status.empty = esc_data.esc[i].empty;
  esc_status.unbalanced = esc_data.esc[i].unbalanced;
  esc_status.escDisconnected = esc_data.esc[i].escDisconnected;
  esc_status.temperatureHigh = esc_data.esc[i].temperatureHigh;
  esc_status.reserved = esc_data.esc[i].reserved;
  p->esc_publisher.publish(esc_status);`
5. Ran `catkin build` and `source devel/setup.bash` successfully.
6. After launching the main launch file, I get the following error
`ERRORLOG/1 @ getValue, L315: Topic 0x25 value memory not initialized, return default`
7. Attempting to publish the new topic `esc` I get the following, which doesn't read any data from the drone.
[image]
8. Ran `rostopic hz /dji_sdk/esc`

[image]

erenaud3, 2019-10-15:
Hi !

Could you share the files you modified here so we can try out what you implemented ? (or even better, a git repo if it is possible to you)

But one thing come to my mind: are you sure that your drone is publishing ESC data on TOPIC_ESC_DATA ? i.e. Did you read it somewhere ine the documentation ?

dji-dev, 2019-10-15:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28846'>Zendesk ticket #28846</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

In order to better understand and help you solve the problem, need to know more information as follows:
1. OSDK-ROS version?
2. The system version and ROS version that you are using?
3. M210 specific model and firmware version
4. About step 4, which file did you add the code to?

But one thing come to my mind: are you sure that your drone is publishing ESC data on TOPIC_ESC_DATA ? i.e. Did you read it somewhere ine the documentation ?
==> A code example is available in the Linux telemetry sample, https://github.com/dji-sdk/Onboard-SDK/blob/master/sample/platform/linux/telemetry/telemetry_sample.cpp#L330

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

ben-isaac, 2019-10-15:
> Hi !
> 
> Could you share the files you modified here so we can try out what you implemented ? (or even better, a git repo if it is possible to you)
> 
> But one thing come to my mind: are you sure that your drone is publishing ESC data on TOPIC_ESC_DATA ? i.e. Did you read it somewhere on the documentation ?

Hi there, thanks for your response. I found the following documentation regarding the ESC data.
https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Telemetry_1_1ESCStatusIndividual.html#details

The files directory are listed below for each step.
Step1: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/include/dji_sdk/dji_sdk_node.h#L388
Step2: Added a new file ESCStatus.msg as described in the original post in the following directory https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.8/dji_sdk/msg

Step3: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/CMakeLists.txt#L81

Step4: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node.cpp#L230

Step5: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L462

ben-isaac, 2019-10-15:
> Public comment from Bruce Cheung in [Zendesk ticket #28846](https://djisdksupport.zendesk.com/agent/tickets/28846):
> 
> Dear Customer,
> 
> Thank you for contacting DJI.
> 
> In order to better understand and help you solve the problem, need to know more information as follows:
> 
> 1. OSDK-ROS version?
> 2. The system version and ROS version that you are using?
> 3. M210 specific model and firmware version
> 4. About step 4, which file did you add the code to?
> 
> But one thing come to my mind: are you sure that your drone is publishing ESC data on TOPIC_ESC_DATA ? i.e. Did you read it somewhere ine the documentation ?
> ==> A code example is available in the Linux telemetry sample, https://github.com/dji-sdk/Onboard-SDK/blob/master/sample/platform/linux/telemetry/telemetry_sample.cpp#L330
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

Hi there,

Thanks for your response. Please see my answers below.

> 1. OSDK-ROS version? **3.7**
> 2. The system version and ROS version that you are using? **Ubuntu 16.04 LTS, ROS Kinetic**
> 3. M210 specific model and firmware version: **Matrice 210 RTK with latest firmware.**
> 4. About step 4, which file did you add the code to? **Please see my previous response to erenaud3's questions above.**

Magic-wei, 2019-10-16:
@ben-isaac Hi Ben,

I noticed that your issue came up with topic `TOPIC_ESC_DATA` memory not initialized, and topics that are subscribed at 50Hz are initialized [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node.cpp#L487) with a vector `TopicList50Hz` as input.

So I think you might need one more step (probably between your step 4 and 5) to add `TOPIC_ESC_DATA` to the `TopicList50Hz` [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node.cpp#L464) in `dji_sdk_node.cpp` before the initialization. It should work.

> Hi there, thanks for your response. I found the following documentation regarding the ESC data. [...]

ben-isaac, 2019-10-17:
> @ben-isaac Hi Ben,
>
> I noticed that your issue came up with topic `TOPIC_ESC_DATA` memory not initialized, and topics that are subscribed at 50Hz are initialized [...]

Hi Wei,
That solved it. Thank you!

dji-dev, 2019-10-17:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28846'>Zendesk ticket #28846</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to have kept you waiting, congratulations on solving the problem.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lid747, 2020-02-28:
Hello everyone,

I have been trying to obtain the same data. I followed all the steps given but at the time of running catkin_make , I get this error:
[image]

I am using ROS Melodic on Ubuntu 18.04LTS. I am able to connect to the M100, M600 and M210 without any problem.

I'd really appreciate the help.

Regards

erenaud3, 2020-02-28:
Hard to say without your code. But it looks like you forget to include ESCData (either in your C++ file, as a .h, or In CMakesList.txt and/or packages.xml)

lid747, 2020-02-29:
> Hard to say without your code. But it looks like you forgot to include ESCData (either in your C++ file, as a .h, or In CMakesList.txt and/or packages.xml)

Hello erenaud3

I created the ESCStatus.msg as said in the first comment

In my CMakeList.txt I added:
`ESCStatus.msg` on the line 69

Then, on my dji_sdk_node_publisher.cpp I added:
`int i = 3; Telemetry::TypeMap<Telemetry::TOPIC_ESC_DATA>::type esc_data = vehicle->subscribe->getValue<Telemetry::TOPIC_ESC_DATA>(); dji_sdk::ESCStatus esc_status; esc_status.header.stamp = msg_time; esc_status.current = esc_data.esc[i].current; esc_status.speed = esc_data.esc[i].speed; esc_status.voltage = esc_data.esc[i].voltage; esc_status.temperature = esc_data.esc[i].temperature; esc_status.stall = esc_data.esc[i].stall; esc_status.empty = esc_data.esc[i].empty; esc_status.unbalanced = esc_data.esc[i].unbalanced; esc_status.escDisconnected = esc_data.esc[i].escDisconnected; esc_status.temperatureHigh = esc_data.esc[i].temperatureHigh; esc_status.reserved = esc_data.esc[i].reserved; p->esc_publisher.publish(esc_status);` in the line 462

In my dji_sdk_node.h I added:
`ros::Publisher esc_publisher;` in the line 388

I included

> @ben-isaac Hi Ben,
>
> I noticed that your issue came up with topic TOPIC_ESC_DATA memory not initialized, and topics >that are subscribed at 50Hz are initialized here with a vector TopicList50Hz as [...]

Thank you very much for your quick response

rikba, 2020-09-07:
We tried the same approach for the DJI M600 Pro but it seems like the ESC message is not published by the autopilot. Is this correct? Did anyone manage to readout the ESC data?

Firmware version: 1.0.1.67
Onboard-SDK-ROS: 3.8.1 + additional ESC status publisher
Onboard-SDK: 3.8.1
Related issue: https://github.com/dji-sdk/Onboard-SDK/issues/627

```
STATUS/1 @ init, L55: Attempting to open device /dev/ttyS5 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0670134540

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x302290E

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: [ INFO] [1599495913.875085714]: drone activated
[ INFO] [1599495913.970497281]: Use data subscription to get telemetry data!
[ INFO] [1599495913.970579545]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 2 result: 0.
STATUS/1 @ startPackage, L352: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L350: Start package 1 result: 34.
STATUS/1 @ startPackage, L352: Package 1 info: freq=50, nTopics=11.

STATUS/1 @ getErrorCodeMessage, L656: startPackage
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_UID

STATUS/1 @ getErrorCodeMessage, L656: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_PACKAGE_DOES_NOT_EXIST
[ERROR] [1599495914.047275117]: Failed to start 50Hz package
[ERROR] [1599495914.047394705]: initPublisher failed
```

dji-kyle, 2021-07-08:
TOPIC_ESC_DATA not support M600
[image]

closed 2021-07-08 by dji-kyle

tkazik, 2021-07-09:
@dji-kyle
It would probably make sense to adapt lines 614/615 then and state it somewhat more explicitely. Also, the M600 had firmware upgrades which date more recent then this code snippet.
Anyway, thanks for confirming that it is not supported.