dji-sdk/Onboard-SDK-ROS #300: Activation failure with M210 and manifold 2C

Ebtiisam, 2019-10-15:
when i try to launch dji_sdk sdk.launch i get this error

<
dji@manifold2:~/Desktop/M210_ws$ roslaunch dji_sdk sdk.launch 
... logging to /home/dji/.ros/log/b1f94636-ef16-11e9-bd30-74da38ebc18f/roslaunch-manifold2-6454.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:42235/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1079440
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 115200
 * /dji_sdk/enc_key: 39d353a8e9ad01b6c...
 * /dji_sdk/serial_name: /dev/ttyTHS0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [6464]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b1f94636-ef16-11e9-bd30-74da38ebc18f
process[rosout-1]: started with pid [6477]
started core service [/rosout]
process[dji_sdk-2]: started with pid [6487]

STATUS/1 @ init, L56: Attempting to open device /dev/ttyTHS0 with baudrate 115200...

ERRORLOG/1 @ _serialOpen, L181: cannot open device /dev/ttyTHS0

ERRORLOG/1 @ init, L61: ...Failed to start serial

ERRORLOG/1 @ sendData, L423: Port closed.
ERRORLOG/1 @ sendData, L427: Open Protocol cmd send failed, send_len: -1 packet_len: 19

ERRORLOG/1 @ getDroneVersion, L1503: Drone version not obtained! Please do not proceed.
Possible reasons:
    Serial port connection:
        * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
        * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
        * TX and RX pins are inverted.
        * Serial port is occupied by another program.
        * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1387: Unable to initialize some vehicle components![ERROR] [1571121642.979869864]: drone activation error
[ERROR] [1571121642.980001324]: Vehicle initialization failed

>

so now what shold i use for the connection since i'm using manifold 2c which only have UART0 and N/A, it is recomended to use USB-TTL but what baud rate and serial name should i put?

erenaud3, 2019-10-15:
Hi !

I am not quiet sure you can perform activation with the manifold. But I am really far from being a DJI expert !

You should definitely use USB-TTL for activation. The baud rate for M210 is 230400 (as mentioned in point 2 in [Read.me.md](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/ReadMe.md)). For the serial port, this command should be enough for Linux:
```
dmesg | grep tty
```
dmesg: display system message.
grep: search for text "tty"

Hopefully you should see some lines with "/dev/ttyUSB0".

PS: I have one advice since I recently perform activation as well. Double check which OSDK version you got with the command : 
```
git clone https://github.com/dji-sdk/Onboard-SDK
``` 
If you didn't checkout to branch 3.8, you certainly got 3.9 version. This will cause errors when you will try to perfom activation.

pavankumarbn, 2019-10-16:
Hi, @Ebtiisam I hope you resolved the problem. Just in case https://pavankumarbn.files.wordpress.com/2018/05/how-to-activate-matrice-100-drone-with-intel-nuc-as-onboard-computer.pdf

dji-dev, 2019-10-16:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28865'>Zendesk ticket #28865</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

 It is recommended that you can refer to the description of UART Ports in the user manual of DJI Manifold2. https://dl.djicdn.com/downloads/manifold-2/20190528/Manifold_2_User_Guide_v1.0_EN.pdf
The DJI manifold2-C UART0(/dev/ttyTHS0) cannot be used to connect to the API interface of M210, so it recommended that you use a USB-TTL device.
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle