dji-sdk/Onboard-SDK-ROS #301: demo_time_sync does not work

maxhofidrone, 2019-10-15:
I have a problem to run the demo program demo_time_sync. The subscription starts successfully on the  dji_sdk node, but no outputs are given at all at the terminal where I start "rosrun dji_sdk_demo demo_time_sync". Does anyone has the same experience? As I understand the code of demo_time_sync.cpp I expect the different times as output of ROS_INFO.
As background information:
I try to subscribe to the GPSUTC time as I need to synchronize observations of the M600Pro with ground base instruments with the drones own GPS/RTK observations. Therefore I'm writing a sort of data logger which should save the different position solutions together with their UTC time stamp.

erenaud3, 2019-10-15:
Hi there, 

I think demo_time_sync can't work unless you have implemented some kind of [hardware synchronization](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-hardware-sync.html). I am not even sure you can run it with a M600pro since [this demo page](https://developer.dji.com/onboard-sdk/documentation/sample-doc/time-sync.html) seems to be about M210.

maxhofidrone, 2019-10-16:
Hi,
thanks for your answer @erenaud3 ! But onboard hardware synchronization is not exactly what I am looking for. There will be no direct synchronization between my ground based instruments and the drone. Basically I'll get two trajectories of the drone. One by the instruments and one by the drone. In a post processing step I want to compare these trajectories. To achieve this I need to save the GPS measurements by the drone with a GPS or UTC time stamp. As any GPS observation basically already include this information, I should be able to access it with the drones own antennas. As I found [here](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Telemetry_1_1PositionTimeStamp.html) it is somehow saved by the DJI OSDK. But I do not know how to access/subscribe to it. Does anyone have an idea how to subscribe to this information?

erenaud3, 2019-10-16:
Hi,

If you take a look at [dji_sdk_node.cpp](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node.cpp) at line 409, you can see this: 
```cpp

  if (vehicle->hardSync)
  {
    vehicle->hardSync->subscribeNMEAMsgs(NMEACallback, this);
    vehicle->hardSync->subscribeUTCTime(GPSUTCTimeCallback, this);
    vehicle->hardSync->subscribeFCTimeInUTCRef(FCTimeInUTCCallback, this);
    vehicle->hardSync->subscribePPSSource(PPSSourceCallback, this);
  }
  return true;

```
Obviously, vehicle->hardSync becomes defined when you set hardware synchronization. That's why I came to the conclusion that demo_time_sync can't be run without hardware synchronization. But honestly, I still don't understand why we need it though. Because as you said, we should be able to get UTC time stamp without being synchronized with GPS.

Anyway, your problem reminds me a lot another [recent issue in this git](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/299). In that one, OP wants to have access to some data not exposed yet to ROS: ESC data. The involved data have type [ESCStatusIndividual](https://developer.dji.com/onboard-api-reference/structDJI_1_1OSDK_1_1Telemetry_1_1ESCStatusIndividual.html#details) and are published over [TOPIC_ESC_DATA](https://developer.dji.com/onboard-api-reference/group__telem.html#ga563d2333dc43253129a3915970f99cfaa1af1640315a760bf79fc05502473e6f9)

In your case, we would need to get some data with typeuint32_t  on the FC's topic  [TOPIC_GPS_TIME](https://developer.dji.com/onboard-api-reference/group__telem.html#ga563d2333dc43253129a3915970f99cfaa119f1c518dfa4145801ed2b47098e3f1).

dji-dev, 2019-10-16:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28872'>Zendesk ticket #28872</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that dji_sdk_demo demo_time_sync currently only supports M210 V2, not M600 Pro.
You can refer to the OSDK development documentation, https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-time-sync.html

As I found here it is somehow saved by the DJI OSDK. But I do not know how to access/subscribe to it. Does anyone have an idea how to subscribe to this information?
==> It is recommended that you use TOPIC_GPS_DATE, but this topic is not implemented in ROS, so you can develop and implementation by yourself.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-10-16:
Thank you very much for your help and suggestions! I will now try to implement this topic by myself.
Best regards,

Max

erenaud3, 2019-10-16:
> Thank you very much for your help and suggestions! I will now try to implement this topic by myself.
> Best regards,
> 
> Max

May I ask you to share what you will do here ? I may need to retrieve those data as well.

maxhofidrone, 2019-10-17:
Yes sure! I'll post it as soon as I have some results

dji-dev, 2019-10-17:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28872'>Zendesk ticket #28872</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It's our pleasure to answer your questions, good luck with your project.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-10-17:
I could publish the GPS_date and GPS_time according the steps mentioned in #299 . There is no need to initialize memory for the data as it is already done in dji_sdk_node.cpp line 536: `topicList5hz.push_back(Telemetry::TOPIC_GPS_DATE);`

But unfortunately the published gps_time is not usable for me. As I could already have recognize earlier it is provided as uint_32 as hhmmss - I do not see the possibility to access the subseconds to synchronize my time series accurate enough. 
Now I will try to access the NMEA message directly to receive a valid timestamp. Does anybody has some experience with accessing these NMEA messages on a M600pro without the hardware sync (M210 only)?

dji-dev, 2019-10-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28872'>Zendesk ticket #28872</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

DJI OSDK provides getTimeStamp() API, which can realize the basic time synchronization of DJI aircraft and ROS system.
Can you describe the specific requirements and application scenarios in detail?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-10-22:
Dear @dji-dev 
as far as I understand getTimeStamp() will provide the POSIX/UNIX time of the onboard computer which is running ROS. This would be enough for synchronizing all kind of data/sensors attached on this computer. In my case I want to track the flight path of the drone (controlled with GPS/RTK) with a ground based geodetic multi-station. The measurements of the multi-stations will include a GPS time-stamp. Therefore I would like to have also the GPS time-stamps on my logging data of the drone. Converting the onboard computers time to GPS time would not be a big deal, but as the onboard computer does not have a realtime hardware clock nor an internet connection, I was thinking of synchronizing the time based on the nmea messages. 

I was trying to access the published nmea massages, but was not successful. Is there any change to access these messages, as they are also used within DJI flight controller? I'm using the M600 so I can not use the hardware sync option.

Best regards,

Max

dji-dev, 2019-10-23:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28872'>Zendesk ticket #28872</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that the DJI OSDK does not support M600 Pro to directly obtain NMEA data (currently only supports M210 to obtain NMEA data using Time Sync function), but DJI OSDK provides multiple APIs to obtain NMEA parsed data, such as:
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/9YMIhX7QdMKmjBuWksWzjJuQ9' target='zd_attach'>inline257608916.png</a>

maxhofidrone, 2019-10-28:
Dear @dji-dev ,
as I'm not able to receive the sub-seconds of `DJI::OSDK::Telemetry::TOPIC_GPS_TIME` [(here)](https://developer.dji.com/onboard-api-reference/group__telem.html#ga563d2333dc43253129a3915970f99cfaa38e3a470ca22cc9eb619cd2181598544), the NMEA data accessible with the M600 is insufficient for my time synchronization purposes. It would be very nice if the NMEA data (and PPS data) could also be accessible with the OSDK on M600 in a future version.

Best regards,

Max

closed 2019-10-28 by maxhofidrone

dji-dev, 2019-10-29:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28872'>Zendesk ticket #28872</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry for the inconvenience caused to you. If you need to use PPS and NMEA data on M600, It is recommended that you click on the links on the right side of the fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team