dji-sdk/Onboard-SDK-ROS #302: DJI Assistant 2 Simulator not starting

AbhishekGS-MW, 2019-10-16:
I am not able to start the Simulator from the DJI Assistant 2 software. I have been using it for a while, and now suddenly the Start button is not responsive. Please let me know how to proceed.

The version of the DJI Assistant 2 is - V2.0.3

I have tried restarting the PC also, but it doesn't help. 

Below is the link to the screen capture video.

https://drive.google.com/file/d/1w3YbpdzwMzkYK6Hi30KlhCGzSse3vKS_/view?usp=sharing

[image]

erenaud3, 2019-10-16:
Hi !

I don't own a N3, I am using a M600pro. But from what I am seeing, your simulation seems to work just fine. Are you expecting your drone to take off when you press Start ? If yes, that's not how that works: you must use the remote controller for it.

dji-dev, 2019-10-16:
Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/28891'>Zendesk ticket #28891</a>:<br><br>Hi! It is recommended that you send email to DJI after-sales technical support for this issue. They are more suitable to help you with it.

closed 2021-07-08 by dji-kyle