dji-sdk/Onboard-SDK-ROS #303: Get waypoint from a third party app

erenaud3, 2019-10-16:
Hi !

I would like to know if there is a way of getting waypoints set from a third party app ?

Exact use case: I launch a simulation thanks to PC Assistant 2 and use Altizure to generate a mission (i.e. a set of waypoints). The drone behaves just as expected but I am not seeing anything on ROS services. When I type the command:
```
rosservice call /dji_sdk/mission_waypoint_getInfo
```

I got:
```
ERROR: service [/dji_sdk/mission_waypoint_getInfo] responded with an error: 
```
and this error in roscore terminal:
```
[ERROR] [1571219280.536870982]: no waypoint mission initiated 
```

So, would it be possible to retrieve the waypoints computed by Altizur on ROS ?

dji-dev, 2019-10-17:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28919'>Zendesk ticket #28919</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry, we have not tested using DJI OSDK-ROS to subscribe to mission information while using a third party APP to generate a mission.
This requires further confirmation and verification, but more information is needed:
1. The Specific model of DJI product
2. Development platform and ROS version
3. OSDK - ROS version

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-17:
To be fair, I only tested with Altizure. Maybe it will work with other applications.

I already put those information at bottom of first message, but here they are again:

**Drone: M600pro
Firmware: 1.0.1.67
System and ROS Version: Ubuntu 16.04 and ROS Kinetic
OSDK version: 3.8.1**

dji-dev, 2019-10-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28919'>Zendesk ticket #28919</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the information you provided. We will make further analysis and verification of the problems you raised and will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-23:
https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp#L240-L255

Okay guys, the problem comes from here. This condition is faulty. Indeed, this function is going to check if some waypoints were already defined in OnBoard-SDK (in the "vehicle" property) before actually requesting the waypoints to Flight Controller (i.e. downloading waypoints).
That's just a non-sense: that means that ROS node can only download waypoints from drone if and only if we already uploaded some.

erenaud3, 2019-10-25:
Okay this problem was harder than I thought. Indeed, first step is to request `WaypointInitialSettings` from drone. Problem: you can't do that because  `vehicle->missionManager->wpMission` is undefined. My workaround was to create an empty `WaypointMission`, just to request `WaypointInitialSettings`. Then, you know how many  `WaypointSettings` you have to request to drone and it's quite simple from now on. You can find my code (I defined a new service in `Onboard-SDK-ROS/dji_sdk/src/modules/dji_sdk_node_mission_services.cpp`). This service is called `dji_sdk/mission_waypoint_downloadInfo`.

DISCLAIMER: this service is going to update `vehicle->missionManager->wpMission` according to waypoints donwloaded from drone (as well as `vehicle->missionManager->wpMissionVector`). BUT you will end up with duplicated mission in `vehicle->missionManager->wpMissionVector` if you call this service several times while drone is performing current mission. (this shouldn't cause any troubles though).

I would have a love to add some mechanism to avoid adding a mission we already downloaded, but it's not possible as OnBoard-SDK is currently implemented. Indeed, neither `WaypointInitialSettings` nor `WaypointSettings` have some kind of unique identifier. Furthermore, it is not possible to request `vehicle->missionManager->wpMission->info` nor `vehicle->missionManager->wpMission->index`, as those attributes are private and no accessors are implemented.

@dji-dev  Would you say that this contribution deserve a pull request ?

```cpp
bool
DJISDKNode::missionWpDownloadInfoCallback(
  dji_sdk::MissionWpGetInfo::Request&  request,
  dji_sdk::MissionWpGetInfo::Response& response)
{
  ROS_DEBUG("called missionWpDownloadInfoCallback");

  ACK::WayPointInit wayPointInit;

  // creating empty waypoint mission
  WaypointMission * newMission = new WaypointMission(vehicle);

  // we use empty new mission to request waypointInitSettings located in drone.
  wayPointInit = newMission->getWaypointSettings(WAIT_TIMEOUT);

  // wait a little bit for anwser
  // WATCH: useful ?
  sleep(1);

  // analyze reply
  if (ACK::getError(wayPointInit.ack))
  {
    ACK::getErrorCodeMessage(wayPointInit.ack, __func__);
    ROS_WARN("ACK shows errors...");
    delete newMission;
    return false;

  } else {

    // now we can init mission in OSDK thanks to the wpInitData we got
    // reminder:
    // typedef struct WayPointInit
    // {
    //   ErrorCode            ack;
    //   WayPointInitSettings data;
    // } WayPointInit; // pack(1)

    vehicle->missionManager->init(DJI::OSDK::WAYPOINT, WAIT_TIMEOUT, &wayPointInit.data);

    // now, we will be able to do:
    // info = vehicle->missionManager->wpMission->getWaypointSettings(10).data
    // even if no waypoint were set from ROS

    // don't need new mission anymore (can't put it outside this condition. Indeed, "return false" above
    // would make program exit without deleting pointer.)
    delete newMission;
  }

  response.waypoint_task.mission_waypoint.resize(wayPointInit.data.indexNumber);
  response.waypoint_task.velocity_range     = wayPointInit.data.maxVelocity;
  response.waypoint_task.idle_velocity      = wayPointInit.data.idleVelocity;
  response.waypoint_task.action_on_finish   = wayPointInit.data.finishAction;
  response.waypoint_task.mission_exec_times = wayPointInit.data.executiveTimes;
  response.waypoint_task.yaw_mode           = wayPointInit.data.yawMode;
  response.waypoint_task.trace_mode         = wayPointInit.data.traceMode;
  response.waypoint_task.action_on_rc_lost  = wayPointInit.data.RCLostAction;
  response.waypoint_task.gimbal_pitch_mode  = wayPointInit.data.gimbalPitch;

  for (int i=0; i< wayPointInit.data.indexNumber; i++)
  {
    // reminder:
[... 19 lines cut ...]
    response.waypoint_task.mission_waypoint[i].waypoint_action.action_repeat = wpData.actionNumber + (wpData.actionRepeat << 4);

    std::copy(std::begin(wpData.commandList),
              std::end(wpData.commandList),
              response.waypoint_task.mission_waypoint[i].waypoint_action.command_list.begin());

    std::copy(std::begin(wpData.commandParameter),
              std::end(wpData.commandParameter),
              response.waypoint_task.mission_waypoint[i].waypoint_action.command_parameter.begin());

    // We also have to update vehicle->missionManager->wpMission !
    // WARNING: we should NOT use vehicle->missionManager->wpMission->uploadIndexData(...). In deed,
    // this function is going to add the waypoints we just downloaded. So they would be duplicated !
    // Instead, we should use vehicle->missionManager->wpMission->setIndex(...) which only impacts Onboard SDK's back end.

    // REMARK: In practice, we are probably better off without setting vehicle->missionManager->wpMission.
    // Because vehicle->missionManager->wpMission->index is never really used.
    vehicle->missionManager->wpMission->setIndex(&wpData, i);

  }

  // By now, waypoints data in OnBoard SDK back end are syncronized with drone's data !
  return true;
}
```

dji-dev, 2019-10-29:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28919'>Zendesk ticket #28919</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your feedback and solution. This idea looks great, but we need to do further analysis and verification and will inform you as soon as any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-07-10:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28919'>Zendesk ticket #28919</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry for keeping you waiting. OSDK-ROS does not support obtaining waypoint task information set by DJI APP/ third-party APP.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side of the fill in the relevant demand information, https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

jaypatrikar, 2021-03-22:
@erenaud3 I think its a good addition. I repurposed it to stop missions from the ROS-OSDK that were started by third-party apps.

closed 2021-07-08 by dji-kyle

Jeff-Vision, 2022-11-07:
Is there a way this can be done on the Linux version of the OSDK? Trying to follow the same logic with the fuction
`WaypointMission::uploadIndexData(WayPointSettings* data, VehicleCallBack callback, UserData userData)`